EdgeTX/edgetx #3911: Flysky EL18 USB always connect even without connected cable

Schnuppi12, 2023-08-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I am turning on the EL18, even without connected USB-C cable, after the start screen, I am getting asked, which connection I want (HID, SD, VCP ) Also the symbol is shown in the top bar.
When I am turning it off, it is asking me, if I want to shutdown, because the EL18 is thinking that the USB-C cable is connected.

**This is happening since 2.8.0.**

When I am entering the bootloader with the actual firmware 2.9-rc2, I can`t enter the menu, because even there, it is saying I have the USB-C cable connected. And I can't navigate there
With the 2.8.2 firmware, I could at least enter die bootloader withou problems to flash the firmware, without pushing the DFU button.

I am not sure, if it is related to this
https://github.com/EdgeTX/edgetx/issues/3296#issue-1609636170
https://github.com/EdgeTX/edgetx/issues/2725#issue-1446804539

Even when I am starting with the USB cable connected and then removing it, my EL18 isn't recoqnising that I removed it.
Since I couldn't find here or somehwere else a similar behaviour, I am not sure if this is a firmware problem or if my EL18 is somehow damaged.

Beside this, everything else is working fine without any problems.

### Expected Behavior

The EL18 should recoqnise if a USB cable is connected or not and shown only then the menu to choose what kind of connection between it and the computer should be established.
Also the bootloader should be entered without any problems.

### Steps To Reproduce

Starting the EL18 without connected USB cable.

### Version

2.9.0

### Transmitter

Flysky EL18

### Operating System (OS)

Windows

richardclli, 2023-08-07:
I believed this may related to hardware issue.  When did you buy your EL18?

Schnuppi12, 2023-08-07:
Hello richardclli, I bought it this year in April. And I was already fearing that it will maybe a hardware issue, since I couldn't find any other person with this specific problem.
I didn't disassembled the EL18 yet. Even I wanted to install the better gimbals. But I am more then happy with the ones installed.
So I will consider buying a new one and using this only for simulator flying.
Thank you for your fast anwser mate.

Schnuppi12, 2023-08-07:
Btw, I just have been at a store to get a new EL18 because it was available there luckily. And when I am comparing them at home now, the new one is more like a frosted red and the old one partly shiny red. Similar what John is saying in his video at RC Video Reviews. Also that the corners don't have the full color spray and you can see the plastic shining through the paint.

**RC Video Reviews:** **Watch BEFORE you buy FlySky Elysium EL-18 with EdgeTX • Important Hardware Info**
https://youtu.be/DHrxQhOF_nk

So maybe I got lucky to get one from the first batch from flysky :D
Anyway, I love it and it is a great piece of equipment for me.

And thank you guys, to bring EdgeTX to it.

richardclli, 2023-08-17:
Yes, the first batch uses electroplating technology.  However, as the casing design of EL18 has quite a lot of sharp corners, it is hard for that part to attract charged paint particles to attach to it.  And that is why the finishing cannot be 100% perfect.

And that is why they switched to paint spraying technology, not that shiny, but can be more perfectly colored.

Did you have a way to send back you problematic unit for repairing?  Or I can help to check if this is a known issue with self repairing know-hows.

closed 2023-09-22 by raphaelcoeffic