EdgeTX/edgetx #3912: Companion 2.8.5+: 6-character sound file names restriction

RipperDrone, 2023-08-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

editor does not allow longer filenames than 6 characters anymore. Hasn't been an issue with OpenTX, now converted to EdgeTX and found that the Companion editor does not accept filenames longer than 6 characters. 

### Expected Behavior

accept at least 8+3 characters filenames

### Steps To Reproduce

define longer filename for sound, try to use e.g. in remote control global settings to play sound.

### Version

2.9.0

### Transmitter

FrSky X9D+

### Operating System (OS)

Windows

elecpower, 2023-08-06:
The filename length is radio profile dependent ie X9 series is 8 and all others 6. Screenshot below is X9D+.
[image]

RipperDrone, 2023-08-07:
@elecpower Good to know, thank you. Maybe the glitch was caused by copying over an existing X9D+ profile to a BetaFPV LiteRadio 3 Pro profile and work from there. Was strange, though. I started working from the 8ch sound filename, tried to delete the 7th character from 'GPSROff' and making it 'GPSROf'. Instead, the edit field showed 'GPSROfOf' after saving, leaving the editor and entering into edit mode again in Companion 2.9RC2.

elecpower, 2023-08-10:
@RipperDrone I cannot recreate the delete effect through the combinations I have tried though I can imagine it occurring.

However there is still an issue with the model and settings file containing a file name longer than allowed for the radio and copy/cut and pasting longer filenames being displayed though when you edit they get truncated.

elecpower, 2023-08-12:
#3929 will fix the copy/cut paste issue

RipperDrone, 2023-08-19:
@elecpower I have re-visited this issue. When I did the copy-paste, I had TWO instances of EdgeTX Companion starten (Winx64, one instance for Taranis X9D+ source, one instance for BetaFPV LiteRadio 3 Pro target), and drag-dropped models from one window to the other. This seems to override the 'profile protection mechanism', and copied models can create some issues, e.g. wrong sound filenames lengths.
Likewise, when loading a locally stored profile from my hard disk, there is a warning when the matching RC profile has not been selected - and then you can only cancel or accept the mismatch. Wouldn't it be more convenient to look up the matching profile (if there is any) in the radio profile roaster and AUTO-SWITCH to the matching profile, maybe showing a notification dialog after the auto-switch OR giving the option of auto-switching to a matching profile named xyz, then letting the user decide y/n/cancel?

elecpower, 2023-08-19:
> @elecpower I have re-visited this issue. When I did the copy-paste, I had TWO instances of EdgeTX Companion starten (Winx64, one instance for Taranis X9D+ source, one instance for BetaFPV LiteRadio [...]

Protection when running multiple instances of Companion each with a different and potentially incompatible profile requires a major rewrite to the copy/paste drag/drop code as is was not considered in the original OpenTX design that EdgeTX inherited.

> Likewise, when loading a locally stored profile from my hard disk, there is a warning when the matching RC profile has not been selected - and then you can only cancel or accept the mismatch. [...]

The current different current profile to settings file on open and needing to cancel, change profile and re-open rather than auto switch maybe not ideal. However there is a lot going on behind the scenes when you change profiles and changing at the point the incompatible message pops up is not practical (major redesign). Also you can set up multiple profiles for the same radio so that would also need to be catered for.

The two PRs generated from your issue fix within one running instance so thankyou for reporting. I'll create a new issue about running multiple instances of Companion as that is a far bigger issue and it needs to be brought to everyone's attention.

RipperDrone, 2023-08-20:
@elecpower thanks 4 ur comments - yes, I was assuming that managing multi-instance operation requires major changes and will be unlikely to happen. Anyway, for copying models from one RC to another (even with post-paste manual corrections needed) seemed the easiest option for me to get my secondary RC into operation on the same models, using Multibind on my copters.

Re auto-match profile, I now understand - thanks 4 explaining the background. One thought, though: Would it also require a major redesign of the 'file loader' and 'profile selection dialog caller' if EdgeTX only checked whether there is a non-matching profile currently selected (which it alrdy does) and then immediately branch over to showing the profile selection dialog, and only after the manual selection continue to check whether it now matches, and if it does load the file? This would not require an auto-switch ans still leave it to the user which profile to pick even in a case of multiple profiles per RC, as you mentioned. Anyway, I acknowledge that many of these proposed changes might be a lot of work, and my judgement of code implications is still very weak. Therefore, I would first of all like to thank you and the team for your continued efforts and great piece of software you have created! Now thinking about it, I'll be heading over to the donation page... :-)

PR #3930 (2023-09-04): fix(cpn): custom function filename length
Fixes #3912
closed 2023-09-04 by pfeerick