EdgeTX/edgetx #3913: Screen control lost while cycling through Themes 2.9R2

Charlevia, 2023-08-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

RM Tx16s 2.9-RC2
Cycling through Themes, opening and closing until RCVR High contrast.
Screen control lost, page button when pressed cycled pic of model, 3 white dots, black rectangle. Shut down system, rebooted, shutdown and removed battery, installed 8.3, then re-installed 2.9, no recovery.
Started pushing buttons, found that both MDL and Page button pressed recovered the system from the scrambled screen

### Expected Behavior

Themes should have changed.

### Steps To Reproduce

Described above

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Linux

Charlevia, 2023-08-08:
Solved by re-installing 2.7
Installing 2.85 did not fix.
2.9 now installed and fine.

closed 2023-08-08 by Charlevia

pfeerick, 2023-08-08:
Did the radio lock up (requiring the battery removed) or crash into EM mode, or did some (most) screen elements simply go black? Did you update the THEMES folder on the SD card at all before updating to 2.9?  If you didn't, you may want to grab it from https://github.com/EdgeTX/themes/releases, as you may have run into an issue with regard to whitespace in the themes file messing up the UI colors. 

*Although that doesn't explain why 2.8.5 would have problems with that theme / cycling through themes* 🤔