EdgeTX/edgetx #3917: Realflight doesnt recognize as joysticks

rooster1973, 2023-08-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Plug in and Realflight does not see the transmitter as a controller that is usable. tried other controller brands and types in setup with no change. previous release works fine.

### Expected Behavior

To detect transmitter as controller

### Steps To Reproduce

Plug in start up Realflight and select controller.

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

rooster1973, 2023-08-09:
> Is that Realflight under Windows or macOS?

Sorry, under windows 11.

Justin

raphaelcoeffic, 2023-08-09:
> Sorry, under windows 11

It seems RF needs some special adapter. Did it work with earlier EdgeTX versions?

raphaelcoeffic, 2023-08-09:
This suggests it should work out of the box: https://steamcommunity.com/app/1070820/discussions/0/2966145784730214965/

```
As long as Windows recognizes your device as a USB game controller it will work in Realflight.
The need for a special dongle has been removed around version 8.
```

3djc, 2023-08-09:
Tested 2.9RC2 on tx16s, i get proper joystick movement under win11 josytick manager, I don't have Realflight to test with however

rotorman, 2023-08-09:
@rooster1973 Press Win+R, enter `joy.cpl` and press Enter. A joystick dialog should open. Select the EdgeTX radio joystick and click on Properties. Do you see sliders move when you move the controls on the radio?

pfeerick, 2023-08-09:
With Realflight Evolution, on Windows 11, "gamepad contollers" control panel detects my handset (running a 2.9-rc) correctly 
[image]
and shows movement for the sticks and configured buttons. 
[image]

Normally it will show a few more axes and buttons, but I'm using the new "advanced" mode to be able to configure buttons and switches to make Realflight work nicer with a radio handset.

3djc, 2023-08-09:
My windows with default settings (deleted radio and models) show a ton more inputs

pfeerick, 2023-08-14:
1) Are you able to launch Realflight with the TX16S connected, and in joystick mode? (as you can't even start Realfight without joystick/controller connected). 
[image]

As this is otherwise working for me on TX16S with 2.9-RC2, with RF Evolution and Windows 11. :shrug:

2) With RealFlight running, after going into Settings -> Controller -> Configure Controller, what happens if you select "Blank Profile" and then "Calibrate"? Is movement detected? If it is, it is simply a matter of editing the profile, mapping the various functions to your preferred controller inputs, and then saving the profile... perhaps as "RM TX16S"?

rooster1973, 2023-08-14:
I'll have to follow your directions and see if it works for me. I will have to update to rc 2 first.

Justin

rooster1973, 2023-08-14:
Rc2 seems to work fine. Although im unable to setup switches for some reason. Doesn't recognize anything but main axis.

Justin

pfeerick, 2023-08-14:
How many mixer lines do you have? You'll only have the controls made
available via the mixer (regardless of whether using classic or advanced
joystick mode).

rooster1973, 2023-08-15:
I don't know I'm new to this radio and software and not sure what you're referring to.

Justin

rotorman, 2023-08-15:
Open model settings and navigate to mixer tab.
If you are new to EdgeTX, it is advisable to work through following: https://edgetx.gitbook.io/edgetx-user-manual/