EdgeTX/edgetx #3919: Switch to emergency mode on tx16s shutdown without having stopped the lua script.

Salto42, 2023-08-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I stop the TX16S without having stopped Log Viewer, the stop logo is displayed, then the radio goes into emergency mode and I can press Power to stop the TX.
This fault did not occur in ETX 2.8.4.

### Expected Behavior

Normal shutdown.

### Steps To Reproduce

Use Log Viewer then stop the TX16S without quitting Log Viewer.

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

pfeerick, 2023-08-10:
I hit a similar problem to this myself yesterday, but via shutdown with
telemetry active. if a telemetry receiver was on, get the warning that a
receiver was still active, and when I then power down the receiver it tells
me telemetry is lost and then triggers EM instead of power off. With a
little bit of garbling of the audio at the end... i.e. as if the crash
triggering EM is happening as the audio is finishing playback.

3djc, 2023-08-10:
Just tried 2.10, don't seem to be able to replicate using @pfeerick method

3djc, 2023-08-10:
Tried 2.9RC2, could not repliacte either

pfeerick, 2023-08-10:
It may actually have been Lua related also... When it was crashing in my
case, it was with MPM (Frsky) TX16S and I had been trying out the new SxR
calibration/config scripts Offer has been working on, and could have left
the Lua open when powering the radio off. I've now been able to get my ELRS
TX16S to consistently EM by loading Roberts FM2M toolbox (standalone lua),
and powering down (with no telemetry link), and it EMs after the shutdown
animation. Not with the ELRS or Betaflight Lua open, but it *will *with the
wizard loader lua open... so maybe graphics related? 2.9.0-RC2 in all
cases.

Salto42, 2023-08-10:
Still the same problem with ETX 2.10.
My internal module is ISRM.

sdellava, 2023-08-22:
Hi team, I've a similar issue that happen even without a running LUA.

My radio seems to crash, the screen freeze in a semi dark state and to reset it I need to remove the battery. Then it restart in Emergency mode.

The issue happen reguallary when I turn on and off the AUX1 or AUX 2 Power switch in the system configuration.

raphaelcoeffic, 2023-09-05:
> I've a similar issue that happen even without a running LUA

Then it's not the same issue ;-) This one should have been closed with #3926.

sdellava, 2023-09-05:
> io is that?

Jumper T18 pro

raphaelcoeffic, 2023-09-05:
> Jumper T18 pro

It could be related to a power supply issue (hardware). Does it happen only when something is connected, or does it happen as well when nothing is connected (and thus nothing is actually consuming power)?

closed 2023-09-05 by raphaelcoeffic

sdellava, 2023-09-05:
It happen, hopefully only when I'm in system setup mode. So whitout connection to the rx. So minimal power consumtion

raphaelcoeffic, 2023-09-05:
> It happen, hopefully only when I'm in system setup mode. So whitout connection to the rx. So minimal power consumtion.

This is switching port to a serial port. This serial port is consuming the power, this is what I meant. Whether the radio module is connected or not has almost no influence on power consumption. We've seen some issues with radios on the 5V rail that would cause a glitch in the MCU when something is switched ON. This is unfortunately nothing we can control.

sdellava, 2023-09-05:
> This is unfortunately nothing we can control.

No problem. It happen only when I move the power switch on AUX1/2 configuration. And anyway it seems to have no effect. I've connected a GPS to AUX2 and it is always powered.

PR #3926 (2023-08-15): fix: EM when LUA script is running and radio is turned off
This revert a minor change part of #2994, which cause very bad EM when trying to power off the radio and a LUA script is running (a wizard for example).

At the time of writing, I have not been able to gather why the change was made in #2994, so it may introduce another issue I'm not aware off, but it does fix the EM issue (tested on TX16S and Zorro).

This fixes #3919

While 2.9 is timing wise the most critical, 2.10 needs it too