EdgeTX/edgetx #3924: Special Functions Missing after Insert

jmxp69, 2023-08-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

2.9.0RC-2 Special Functions are defined: SF1, SF2, SF3, SF4, SF5, (skipped 6), SF7, SF8, SF9.

After inserting a new special function and not making any changes--just pressing return, the list of Special Functions does not allow me to scroll down to the new SF10 (due to SF9 being pushed down a row) and after tabbing to a different screen and back, the unedited SF7 is removed from the list and the prior Special Functions are pushed down one slot.

### Expected Behavior

After inserting a new SF, I should see the entire list of special functions simply moved down one row and be able to scroll to all defined SFs.

I should be able to go to different tabs and back and not lose the new S7 even if it just default values.

### Steps To Reproduce

Special Functions are defined: SF1, SF2, SF3, SF4, SF5, (skipped 6), SF7, SF8, SF9.

Press SF7, Insert and the SF7 edit screen pops up with default values.
Press Return, SF7 appears in the SF list with the following:  --- Override - CH1 = 0 (default values).
SF8 (prior SF7) is shown. SF9 (prior SF8) is shown. SF10 (prior SF9) is not shown and I can't scroll to see it.

Tab to Logical Switches and Back to SF - The new SF7 with default values is gone from the list, and the original SF7/8/9 are now displayed as SF8/9/10.

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

2.9.0-RC2

philmoz, 2024-08-04:
Not reproducible in 2.10 or 2.11.

eighty8curls, 2025-01-04:
Also experiencing this bug on v2.10.5. edgetx-tx16s on the RadioMaster TX16S mk1.

pfeerick, 2025-02-14:
This appears to have been resolved in 2.10, not reproducible on TX16S with 2.10.6. There is no need to swich tabs to get the correct list of SF, nor is the newly inserted SF lost.

closed 2025-02-14 by pfeerick