EdgeTX/edgetx #3927: problem in edgetx 2.9-rc 2

dulbehek, 2023-08-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

 brightest set under 50% in  jumper t18 pro speaker will be made a bad noise.  over 50% still have very little noise in the speaker 

### Expected Behavior

I want that noise disappear

### Steps To Reproduce

set brightness under 50%

### Version

2.9.0

### Transmitter

Jumper T18

### Operating System (OS)

Windows

raphaelcoeffic, 2023-08-11:
@dulbehek is that really specific to `2.9.0-rc2`? What about other version? I believe this must be related to some electro-magnetic feedback between traces on the PCB. I don't think this is specific to any version. We haven't changed the PWM frequency or anything like that for the backlight.

3djc, 2023-08-11:
Could be linked to audio mute. Go SYS/Hardware, ensure that "Mute if not sound' is enabled

nightwolfix, 2023-09-02:
> @dulbehek is that really specific to `2.9.0-rc2`? What about other version? I believe this must be related to some electro-magnetic feedback between traces on the PCB. I don't think this is specific to any version. We haven't changed the PWM frequency or anything like that for the backlight.

it is specific to 2.9 and it was not there before. It goes away if you mute the radio and then unmute it from the menu, as suggested by 3djc. I have seen many other people in the forum with the same issue, maybe you should take a look

raphaelcoeffic, 2023-09-03:
@nightwolfix now I see! In previous versions, we would mute automatically on some radios when no sound is used. Now it is based on the menu options, and available on all possible targets (where we can effectively mute sound output).

3djc, 2023-09-03:
But I would not rate this as an issue. Muting introduces audio lag, and that's not good, when most users do not need mute. So default unmuted, and if you are one of users needing it, enabled it.

pfeerick, 2023-09-04:
Agreed. I will mention it in the 'changed behaviours' section of the release notes, as it may catch people unawares like here since their radios might now be noisy in situations where they were quiet before.

closed 2023-09-04 by pfeerick

gnuton, 2023-09-05:
Noticed this issue on my T18 Pro too. I confirm "Mute if no sound" option in the hardware screen works.
The noise is only audibleon my transmitter when the external ELRS gets powered and the brightness is low.

wlgrd, 2024-07-18:
I just bought a new Radiomaster Boxer and upgraded from 2.8.4 which it came with to the latest 2.10.2 and came here because I have the same issue. The humming is lowest on 100% screen brightness but higher the more I turn it down (except from 0 where it is completely gone). Can anyone advice? I'm quite new to this and the PWM sound is quite annoying.