EdgeTX/edgetx #3928: 2.9.0-rc2 Companion missing ADJUST_GVAR special function

rburrow87, 2023-08-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

ADJUST_GVAR does not seem to pull into companion, which is supposed to be SF4 below:
[image]

In the TX (TX16S MKII):
[image]

### Expected Behavior

ADJUST_GVAR SF should populate in Companion

### Steps To Reproduce

Create an Adjust special function like in the above screenshot, then read the model in Companion

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Windows 11 22H2

### Anything else?

Tried my best to search for an existing issue!

elecpower, 2023-08-12:
@rburrow87 works for me with a new file and the sim. Can you please upload a copy of files you are reading into Companion.

elecpower, 2023-08-12:
@rburrow87 please check you Companion radio profile settings to ensure option nogvars is not ticked as I can replicate if it is ticked.

rburrow87, 2023-08-12:
Ahh that's what it was. I thought it might have been the enabled features list causing it to be missing but Global Variables was visible and enabled so I didn't think to check radio profile.

elecpower, 2023-08-12:
Can never have too many places to check lol

closed 2023-08-12 by elecpower