EdgeTX/edgetx #3944: Timer1, 2, 3 missing in logical switches

ParkerEde, 2023-08-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I noticed with the current main that the timers1 2 3 can no longer be selected at V1 for the logical switches, e.g. a=x. They are no longer on the list

### Expected Behavior

Timer1, 2, 3 should be selectable 

### Steps To Reproduce

Open logical switches
Select a=x
try to select Timer1 for V1

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Windows

### Anything else?

Current main

3djc, 2023-08-18:
Have you defined the timer ? It needs to be to be present on the list

ParkerEde, 2023-08-18:
What means defined?
I already have a working logical switch that works with timer3 in etx 2.9. After I updated to the current main the timer3 is no longer there, instead only --

3djc, 2023-08-18:
Defined

[image]

Undefined

[image]

3djc, 2023-08-18:
I didn't develop that change, but to me, it goes in the right direction of hiding what is undefined

ParkerEde, 2023-08-18:
I have in etx 2.9 special functions an entry "Sag value" - Motor (Tmr1).
After the update there is also only "Say value" - --.
After the update this is somehow deleted. But when I open the list, I can select "Motor" for Tmr1.

ParkerEde, 2023-08-18:
> 

is there another way, where I can set a bool var except global variables? GV1-9 is in use for other things.

ParkerEde, 2023-08-18:
okay

closed 2023-08-18 by ParkerEde