EdgeTX/edgetx #3947: EdgeTX 2.9.0 RC2 does not boot on Jumper T-Lite V2 (with F4 MCU) when flashed using Buddy

AndrewBCN, 2023-08-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

EdgeTX 2.8.5 boots fine to UI on Jumper T-Lite V2 (with F4 MCU) (with internal ELRS module), however when I flash EdgeTX 2.9.0 RC2 using Buddy, I cannot boot the transmitter, I get a black screen. Reflashing 2.8.5 using Buddy then boots normally again.
Note that 2.8.5 does not distinguish between Jumper T-Lite variants.
I have the Jumper T-Lite V2 variant that uses an F4 MCU (actually a Chinese clone of the STM32F4).

### Expected Behavior

The expected behavior would be that 2.9.0 RC2 would boot normally on the T-Lite V2 transmitter, just like 2.8.5, when flashed using Buddy.

### Steps To Reproduce

1. Flash Jumper T-Lite V2 transmitter using Buddy to 2.9.0 RC2 (Jumper T-Lite with F4 MCU).
2. Press and keep pressed power button. Nothing happens, screen stays turned off (black).
3. Flash 2.8.5 (note that 2.8.5 does not distinguish between Jumper T-Lite variants).
4. Press and keep pressed power button. Transmitter turns on normally, screen lights up, "Welcome to Edge TX" sound plays, etc.

### Version

2.8.5

### Transmitter

Jumper T-Lite

### Operating System (OS)

Windows

### OS Version

Windows 11 using Chrome

### Anything else?

This could be either Buddy which is not downloading the correct firmware or the "2.9.0 RC2 T-Lite with F4 MCU" firmware has a bug that prevents it from booting normally. Note that 2.8.5 works perfectly fine.

Thank you for taking a look at this when you have some time.

pfeerick, 2023-08-19:
This has already been fixed in the 2.9 branch, but was after RC2. You can either grab the latest build for 2.9  - [currently this one](https://github.com/EdgeTX/edgetx/actions/runs/5862030305)  - or wait for the next RC (or possibly 2.9 release).

closed 2023-08-19 by pfeerick

AndrewBCN, 2023-08-19:
Thank you very much for your answer, I guess I'll wait for the 2.9.0 release, since 2.8.5 is working fine right now.
And as usual many thanks to the entire EdgeTX development team for your fine work.