EdgeTX/edgetx #3948: Mixer slow function does not work reliably

icebreaker-ch, 2023-08-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I operate the flaps by a slider  or set them at specific values for "Butterfly" depending on the positions of a Switch. Although all mixer lines have a slow time of 2 seconds. the flaps sometimes move fast when flipping the switch (see attached screen cast).
A simple configuration to reproduce the problem can be found in the attached files.
[image]
[image]
[Test.yml.txt](https://github.com/EdgeTX/edgetx/files/12384325/Test.yml.txt)
[cinnamon-2023-08-19T002143+0200.webm](https://github.com/EdgeTX/edgetx/assets/3645375/808f3cb2-e5d8-476f-9270-aeae5c6beaa4)

### Expected Behavior

The slow function shall apply on all changes of the output, regardless of the source of input.

### Steps To Reproduce

1. Choose Horus X12S as Radio in the simulator
2. Load the Test-model from the attached file test.yml (remove the .txt extension first)
3. Flip the switch SB down and up
4. Watch the Servo-Output of CH1. Sometimes it is slowed down, sometimes not.

### Version

2.9.0

### Transmitter

FrSky X12

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.2 Cinnamon

philmoz, 2024-08-04:
Slow up and down do not work properly on mix lines that are also controlled by switches.

Try moving the switch control to the input line instead:
[image]
[image]

icebreaker-ch, 2024-08-04:
Thanks for this solution which actually works. In the meantime, I found another solution by using Flight modes. Switching the flight mode in a mixer line also obeys the slow up and down settings as expected.