EdgeTX/edgetx #3950: Companion support radio profile protection for copy/paste and drag'n'drop across multiple instances

elecpower, 2023-08-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When running multiple instances of Companion each can have its own active profile that can potentially be incompatible eg B&W vs color.

Currently there are no profile checks when performing copy/cut and paste or drag'n'drop across instances of Companion potentially resulting in corrupt settings in the destination model.

### Describe the solution you'd like

Profile checking across instances of Companion.

### Describe alternatives you've considered

Only allowing the running of one instance of Companion - expect wrath and anger.

### Additional context

Reported as at Companion 2.9

RipperDrone, 2023-08-20:
@elecpower thank you. I feel there might still be a glitch with file names / lengths. Pls check my OSD video, I have uploaded it to YT for convenience: [https://youtu.be/XGlm9wgxwJk](url)

In this case, I am running 2.8.5, and I have loaded the profile from my BetFPV Radio Lite Pro 3, then edited since I spotted an awkward filename with 'duplicated letters'. Behavior when trying to change the filename now is strange. Not talking sound file names anymore, this is about the craft name...?

RipperDrone, 2023-08-20:
Model and Simulator show name correctly, model directory doesn't. Strange... 
[image]

Might be unrelated to copy-paste, but initially the model names were generated by a copy-paste process...

elecpower, 2023-08-22:
Thanks for the video that really helps.
Without testing I'm almost certain the initial paste creates the platform for the bug to raise its ugly head.