EdgeTX/edgetx #3951: no throttle after a certain sequence of running wizard loader LUA

ror1948, 2023-08-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

bug in WizardLoader.lua or plane.lua
create new model using wizard and answer NO motor
run LUA script Wizard Loader and select plane and say Motor YES
when you test for a throttle output on the Rx , you will  have no throttle but there is output on throttle channel in the output screen

### Expected Behavior

expect there to be throttle output on Rx

### Steps To Reproduce

- go to manage models and select new model
- then select wizard
- select plane
- answer motor  NO
- answer remaining questions any way.
- finish the wizard
Back on main screen on the newly created model
press SYS
run wizard loader LUA
by default it operates on the current selected model
select plane
-answer motor YES
- answer remaining questions any way.
- finish the wizard
bind model to a Rx and you will have NO throttle but you will see the throttle channel move in the output screen

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

2.10.0 nightly b0e64c8d
2023;07;11
2;46;57

3djc, 2023-08-20:
If it moves ok on channel monitor, then LUA did it job just fine, issue is somewhere else. Please share a copy of your SD with the radio set to the model it is happening to.

ror1948, 2023-08-20:
It is very repeatable. I did it on two different radios with different firmware and 2 DSM Rx.Orange and AR610.

3djc, 2023-08-20:
Can I have a picture of your module settings and the mixer screen?

ror1948, 2023-08-20:
my settings and sdcard...had to delete several dir to reduce size
[nothrottleSD.zip](https://github.com/EdgeTX/edgetx/files/12388836/nothrottleSD.zip)
models,radio and scripts are all there
[no-throttle.zip](https://github.com/EdgeTX/edgetx/files/12388842/no-throttle.zip)

model is called nothrottle

3djc, 2023-08-20:
I also need a picture of your module setup please

ror1948, 2023-08-20:
> Can I have a picture of your module settings and the mixer screen?

by module settings do you mean internal RF?

ror1948, 2023-08-20:
4in1 has mm-stm-serial-aetr-v1.3.3.20

3djc, 2023-08-20:
You model works fine here. I don't have any spektrum, so used frsky receivers, but clearly the multi gets what it should, not sure where it breaks after that

ror1948, 2023-08-20:
yes but you changed alot to set that up and rebind to frsky
different mapping on frsky

ror1948, 2023-08-20:
by module settings screen do you mean internal RF screen?

3djc, 2023-08-20:
> yes but you changed alot to set that up and rebind to frsky
> different mapping on frsky

Absolutely not, I used fully your sd, only changes are with multi settings, nothing changes edgetx side between the two. I suggest you try with another spektrum receiver, as I can see nothing wrong with your model.

ror1948, 2023-08-20:
well maybe bug is in the multi side.
Like I said before "It is very repeatable. I did it on two different TX16s radios with different firmware and 2 different DSM Rx. one Orange and one AR610."

it maybe a DSM multi issue

3djc, 2023-08-20:
Unsure, maybe another dev does have a spektrum receiver to try