EdgeTX/edgetx #3952: Companion does not handle timers the way the radio now does

3djc, 2023-08-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Companion offers is list (like SF Play Value) all 3 timers

### Expected Behavior

Only defined timers should be presented, as the radio now does.

### Steps To Reproduce

Create or edit a model;

Add a Play Value special function, check list of possible values

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Linux

elecpower, 2023-08-20:
@3djc on the subject  of consistency, why then does radio (used TX16S libsim) list inactive timers for Set and Reset functions?

3djc, 2023-08-20:
Not only that, but it uses different names in yaml. It's all historical and getting slowly cleaned, but while you are at companion, do it for both, I'll do it radio side too :)

elecpower, 2023-08-20:
Do you want to fix the yaml while we're about it ie Set and Reset def to same as expoData and Play Value?

expoData:
  - srcRaw: TIMER1

customFn:
  0:
    swtch: SA0
    func: SET_TIMER
    def: Tmr1,0,0
  1:
    swtch: SA0
    func: RESET
    def: Tmr1,0
  2:
    swtch: SA0
    func: PLAY_VALUE
    def: TIMER1,1x

elecpower, 2023-08-20:
That's provided TIMERx fits in parameter data structs

3djc, 2023-08-21:
Let me try reading both (for set and reset) and writing TIMERX

3djc, 2023-08-21:
@elecpower all done and working radio side. It's now using only TmrX for yaml writing (but TIMERX is still accepted for play value for downward yaml compatibility)

PR #3956 (2023-08-28): fix(cpn): only timers with active modes should be included in lists
Fixes #3952

Summary of changes:
- add mode filter to source lists
- trigger update to source lists and cross references when mode changed in model edit
- fix mode off test
- add mode filter to Set functions and Reset function parameter
- support new yaml format for timers

MERGE NOTE: partner for #3949