EdgeTX/edgetx #3959: Name for 6 POS switch / simulator GPS default value is displayed incorrectly

ParkerEde, 2023-08-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

with 2.9RC2 the name for the 6POS switch (Horus X10) , as well as in the telemetry simulator the default GPS data is displayed incorrectly.
The 6POS switch is called everywhere : 5x {6P?}
The default GPS data in the telemetry simulator: 5x {25.9973,-97.1572?}

### Expected Behavior

6POS Switch defaultname should be: 6P (like in Radiofirmware)
default GPS data in the telemetry simulator: 25.9973,-97.1572

### Steps To Reproduce

open companion
edit radio setting
observe the name of 6P switch

run Simulator
open telemetry simulator
observe the default GPS Value

### Version

2.9.0

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Windows

### OS Version

Windows 11

### Anything else?

Etx 2.9 rc2

elecpower, 2023-08-23:
In English looks okay

[image]

Simulator telemetry

[image]

elecpower, 2023-08-24:
I checked a couple of other languages (French, Italian, Dutch) and they display 6P so it looks like we have a translation corruption.

PR #3968 (2023-08-26): fix(cpn): de translation
Fixes #3959

[image]
closed 2023-08-26 by pfeerick