EdgeTX/edgetx #3961: Hardware Info TX/RX Screen too small

ParkerEde, 2023-08-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If int. ISRM and ext. R9M2019 ACCESS modules are used and one receiver is bound at a time, the window in the Hardware Tx/Rx Info is too small. The information of the R9MMINI-OTA is no longer displayed. It is also not possible to scroll down. If I then deactivate the ISRM module for a test, there is enough space again and I can then see the Rx information from the R9MMini-OTA.

### Expected Behavior

The Infos screen should be scrollable so that the data can all be displayed.

### Steps To Reproduce

1. Bind 2 ACCESS reiceiver to 2 ACCESS Moduls
2. open Hardware TX/RX Info screen
3. observe, there is not enough space for all Info

### Version

2.9.0

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Windows

### OS Version

Windows 11

### Anything else?

etx 2.9 rc2

philmoz, 2023-08-26:
In 2.9 the dialog should be scrollable; but does not show the scroll bar.
2.10 should already fix the missing scroll bar.

ParkerEde, 2023-08-26:
Scroll with rotary encoder doesn't work with x10

PR #3987 (2023-09-23): fix(color): Allow scrolling of the version info dialog popup with the rotary encoder wheel.
Fixes #3961
closed 2023-09-23 by pfeerick