EdgeTX/edgetx #3975: CURVES effected by last adjustment in INPUTS

doubleO00, 2023-08-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After setting up curves, ie throttle and pitch curves,  you can view a display to the side in the curve screen that reads the position of the throttle to the curve, and the bar that also moves with the throttle input.

if you back out and make any adjustment to Ailerons in the Input screen and go back to curves, the bar and output to the left will no longer move with throttle, but moves with Aileron.

If you then go back to inputs and make any adjustment to throttle the bar and output will now move with trottle

### Expected Behavior

Throttle curve display should not be effected by adjustments to Aileron 

### Steps To Reproduce

Create a throttle curve and pitch curve 
adjust aileron input

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

raphaelcoeffic, 2023-09-05:
What you're seeing is some kind of "auto-control" to show you how the curve behaves. Please note that you don't create explicitly a "Throttle Curve" or a "Pitch Curve". You create 2 curves, and these happen to be used by Throttle and the other by Pitch, but could be used by any other control as well if you would like it to.

doubleO00, 2023-09-05:
Is there a way that in this feature when we create a curve and assign a switch to it that only our assigned switch be shown in the display ?

pfeerick, 2024-04-08:
Actually, yes, there is. But first, a minor explanation of what is happening. If you go into the Curves tab, and edit a curve, *there is no connection between the curve, and what it is used for*. Which is why whatever control you move significantly will be used to show the relationship between the input and output of the curve. i.e. if you move aileron stick, curve preview will react. If you then move elevator stick, curve preview will then react to that. 

Now, if you instead go into an input, go to the curve entry, and long press ENTER on the dropdown field (or long press the dropdown field if you have a touchscreen), the curve editor will come up for the selected curve, but now you've given the context - what control you have it linked to - so will only react to that controls movement ;)

closed 2024-04-08 by pfeerick