EdgeTX/edgetx #3976: Confirmation for duplicate model like there is for delete model

mshonak, 2023-08-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I'm always frustrated when I press on duplicate model by mistake and it duplicate it without asking me for confirmation like it does when I try to delete model . 

### Describe the solution you'd like

It should ask me if I am sure before making a duplicate model or even make the duplicate but give it a new name ( or just add to the new duplicate model name a number ) .

offer-shmuely, 2023-08-27:
I am also find sometimes duplicated models that come from no where, looks like sometime i press the incorrect button, and it duplicates

pfeerick, 2023-08-28:
For what radio, and what ETX version? ColorLCD since 2.8 does have a confirmation on duplicate.

mshonak, 2023-08-28:
> For what radio, and what ETX version? ColorLCD since 2.8 does have a confirmation on duplicate. Duplicating (copy) on B&W is a two step process, so I suspect you're still running an older ETX version on colorlcd.

I am using edge tx 2.8.5 on tx16s  and there is no confirmation when duplicate model .

mshonak, 2023-08-28:
Sorry - I checked and it did ask if i want to duplicate model on my radio TX16s EdgeTX 2.8.5  -  I got this request from friend and didn't check myself this issue,  i just remember that it was duplicate the model without asking  .

closed 2023-08-28 by mshonak