EdgeTX/edgetx #3977: Edit Radio Settings close Companion

TheIsotopes, 2023-08-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

I'm using latest master Companion. Click on "Edit Radio Settings" and Companion will close.

### Expected Behavior

Edit Radio Settings will working.

### Steps To Reproduce

see Current Behavior

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2, Radiomaster Zorro

### Operating System (OS)

Windows

### OS Version

11

elecpower, 2023-08-28:
Confirmed illegal address

TheIsotopes, 2023-08-29:
It works again. thx 👍

PR #3978 (2023-08-29): fix(cpn): Edit radio settings application crash
Fixes #3977

Bug in global functions when building lists containing timer related entries introduced by #3956

Summary of changes:
- test for existence of a model pointer in function parameters before calling model functions