EdgeTX/edgetx #3980: global functions missing

ParkerEde, 2023-08-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

after updating my TX16S from 2.9.0rc2 to the current main, the global functions are missing in SYS.

It is available in the Companion Simulator, only missing in the radio firmware.

### Expected Behavior

global functions should be available

### Steps To Reproduce

1. update radio tx16S to current main from 2.9 rc2
2. observe SYS try open global functions

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Win11

### Anything else?

current main

ParkerEde, 2023-08-29:
my mistake. Close

closed 2023-08-29 by ParkerEde

JDessert, 2023-09-11:
I'm having the same issue with not having Global Functions tab displayed in V2.9.0 production release of Edge TX.  Global Functions I had setup prior to upgrading still function, but the menu for Global Functions are not visible to either edit or delete.

If I go into Enabled Features under System Settings Menu, I can enable/disable the GLOGAL FUNCTIONS tab, but it has no effect on what is displayed.

ParkerEde, 2023-09-11:
In my case, I had deactivated the "global function" for one model before the update because it should not be used there. After the update to 2.9.0, the menu item global functions in the radio settings was also gone. It seems that in 2.9.0 it is now the case that if the menu item is not visible, the global functions are not applied.
Personally, I would have thought that menu visibility and applying of the global functions are configured separately, but this is not the case here.

JDessert, 2023-09-11:
I have Global Functions disabled for all my heli models, but enabled for all other models.  I reflashed with 2.8.5, ensured Global Functions menu was visible and then reapplied V2.9.0 and Global Functions menu seemed to have started working properly again at the system level.

ParkerEde, 2023-09-11:
Therefore, if you select a helicopter, the global functions will not be visible in the radio settings and if you select another model, it will be visible in the radio settings.

JDessert, 2023-09-11:
Exactly.  Just new and unexpected behaviour in V2.9.0 that was different than v2.8.x and older.  Under versions older than v.2.9.0, Global Functions system menu was visible even if it was disabled for a particular model.  Not sure if I like the new way as it makes it appear that Global Functions system menu is broken if you are not aware of the change in display behaviour.

ParkerEde, 2023-09-11:
I also stumbled across it during the test phase and then created this ticket and only noticed afterwards how the connections are. You can certainly argue about which is better. I have come to terms with this and therefore closed the ticket. However, if the pressure and the misunderstanding from the community grows, I'm sure they will take another hand here.