EdgeTX/edgetx #3982: CPN: LS and RS not set correctly after reading Model and settings from radio

ParkerEde, 2023-08-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

in Companion in the hardware settings the sliders LS and RS not set correctly after reading Model and settings from radio

### Expected Behavior

in the Comapnion the settings from the radio profile should be taken over correctly after reading the model and radio settings

### Steps To Reproduce

- build companion with RM TX16S with current main 
- make sure that in the radio under Hardware/Sliders LS and RS is assigned correctly and is no longer set to "none"
- start companion
- read modell and radio settings
- observe Radio Settings/TAB Hardware
- both Slider LS and RS are set to "none"

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Win11

### Anything else?

current main

pfeerick, 2023-08-30:
Thanks for reporting this... I hit it the other day myself when preping
companion support for another radio but wasn't sure yet if I'd screwed up
something in the implementation or not. 😅

3djc, 2023-08-30:
Sound like the return of the vengeance of adc refactor part II

elecpower, 2023-08-30:
and #3870 will completely break it again

3djc, 2023-08-30:
Indeed, no point in fixing before 3870 gets in

elecpower, 2023-08-30:
The Companion refactor will parse the hw json files for 2.10 onwards but as usual the biggest pain point is data mapping from the earlier hard coded definitions.

3djc, 2023-08-30:
We are currently generating json on the fly per radio, with much more info than you need. We probably should try to make one file for companion use that gives all the info for all radio at once

elecpower, 2023-08-30:
It is my understanding that on the fly is only a temporary method.
I have some code that handles a permanent json file per radio and ignores the entries that Companion does not care about.
I'm not in favour of an all radios config file as differences will inevitably occur in maintenance.

PR #4406 (2024-02-04): refactor(cpn): Support JSON radio hardware definitions
Summary of changes:
- remove most of the 1st attempt
- move from static board configurations to use radio definitions json files. This also has the advantage of keeping radio sims, firmware and Companion in sync
- group separate structs for same item eg pots into single structs
- maintain support for reading legacy radio settings configuration formats eg bin, otx, etx and convert all formats inc yml as necessary
- yaml encode and decode
- IMPORTANT: 'cmake --build . --target native' and 'cmake --build native --target hardware_defs' MUST be [... cut]
closed 2024-02-04 by pfeerick