EdgeTX/edgetx #3983: We dont have "Set Main Screen" function in black and white yet

arhifisyant, 2023-08-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

As a user of black and white monochrome screen, we may need to bring the telemetry screen to active using special function. 
In comparison to the color LCD screen, that have ability to "Set Main Screen", currently in 2.9 RC 2, we dont have similar ability. 
https://www.youtube.com/watch?v=5a-sNYZ-7uc&t=12s 

### Describe the solution you'd like

Implement new feature in black and white radio, that have similar ability like "Set Main Screen" in color radio. We may name it "Set Tele Screen", since we dont have customizable main screen. Remember, we only have some preset main screen in black and white and it's not editable. Instead, in black and white, telemetry screen is customizable, and also it has multiple pages. Please make it happen, so the black and white become more equivalent in terms of functionality, compared to the color radio.

Vasili-Sk, 2024-07-12:
Up!

arhifisyant, 2024-07-12:
We're still waiting. Patiently. Thank you in advance

arhifisyant, 2024-08-26:
https://youtu.be/9MgL2QTSVj8?t=486

Bardwell also complain about the telemetry screen on the black and white radio. Overall he mention about the inferior tele 
screen in black in white. Spesifically he mention about he has to click on the tele button to open up the tele screen. "and I have to press that button to get there" Bardwell. This should be imroved. Come on, please

pfeerick, 2024-08-26:
Yeah, had just watching that... the poor darlin'... has to press a button to get the telemetry screen to show... clearly forgot you need to keep pressing return on a colorlcd to get back to main view to do the same... can't see the widgets if you're not at the main view! :rofl: 

This might have a better chance of happening sooner rather than later, given for some B&W radios 2.11 will be the last version that will formally supporting them (i.e. they may still work in subsequent versions, but no promises, and if the firmware for them breaks, it breaks).

arhifisyant, 2024-08-27:
Thank you for the attention. Hopefully this will soon implemented. The idea is is just trigger the radio to show and close the tele screen using logical switch.

slackista, 2024-10-03:
+1
Would love to always see screen1 with telemetry when eg. arm is switched.

arhifisyant, 2024-10-03:
> +1 Would love to always see screen1 with telemetry when eg. arm is switched.

Yes. 1 another vote for this. Thank you

nleach7, 2024-10-03:
+1

It would be a great to be able to automatically switch to FM2M when I arm like I do on my TX16s.

PR #5589 (2024-10-17): feat(bw): add 'Set Screen' special function to B&W radios
Fixes #3983 

The parameter for the 'Set Screen' function can be from 0 to 4.
- when set from 1 - 4 it will load the corresponding telemetry screen (if it is set up).
- when set to 0 it will reload the normal main view.

Needs translation (must be kept short to fit on B&W screen):
```C
#define TR_SF_SET_TELEMSCREEN          "Set Screen"
```
closed 2024-10-17 by pfeerick