EdgeTX/edgetx #3990: Upgrading to edgetx-2.9.0 stuck on tx16s radio

skazi1982, 2023-09-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When i wanted to update the firmware from 2.9.0-rc1 to 2.9.0, the progress bar stopped before the end and nothing happens, 30 minutes passed and still nothing.
[image]

### Expected Behavior

The progress bar should go to the end and it should pop up ready.

### Steps To Reproduce

1. Copy firmware to sd card
2. Started flashing firmware
3. Writing firmware stuck and nothing happens

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

gagarinlg, 2023-09-02:
This sometimes happens and we do not know why, just try again and it should finish.

pfeerick, 2023-09-02:
You'll need to unplug the battery in order to retry it, but it should work the second time.

skazi1982, 2023-09-02:
Ok, thanks. The third time it worked. It's the first time I've had this situation.

closed 2023-09-02 by pfeerick