EdgeTX/edgetx #3992: Power up transmitter without transmitting signal / Disable module at startup

doubleO00, 2023-09-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Power up the transmitter without  transmitting a signal

### Describe the solution you'd like

Would be great if we could power up the transmitter while also holding a specific button to disable module transmission

Sometimes you want to make adjustments to models parameters without sending power to module, I know you can enter settings and disable module, but it would be great to do that at power up with an additional button being pressed

raphaelcoeffic, 2023-09-03:
All modules that we know of can be configured on the fly, and their output is not harmful, so that I'm wondering what you are concerned about. You can just leave the receiver off, change whatever you need to, and switch the receiver on whenever you're ready, right?

CamGenius, 2023-09-10:
This would actually be handy I think. I will often have last flown a model that is on the internal module so I have to screw on the antenna before I power it up even if I am only going to switch to a model that uses an external module.