EdgeTX/edgetx #3995: Wrong visualization of switches on 9XD

MicheleVilla, 2023-09-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On main page switches are visualized wrong like attached picture
Firmware is self compiled (2.9)
On companion simulator the visualization is correct but not in the mode as displayed in the attached image

### Expected Behavior

Correct visualization of all switches

### Steps To Reproduce

go to main page and scroll with PAGE key

### Version

2.9.0

### Transmitter

FrSky X9D

### Operating System (OS)

Linux

### OS Version

Linux Mint 20

### Anything else?

[image]

pfeerick, 2023-09-03:
Is this really 2.9, or is this main (which is 2.10 - as will be indicated on the version page).

MicheleVilla, 2023-09-03:
Screenshot was take in main, but in recompiled firmware from 2.9 the
visualization remains the same
If you want I can post the screenshot from right version (2.9)

pfeerick, 2023-09-03:
No, that's great... means the issue really should be fixed by #3996 as it will be due to the new hardware abstraction layer in 2.10 / main. Thanks for reporting it! :)

PR #3996 (2023-09-29): fix(b&w): Fix radio UI display for switches on X9D, X9D+ and X9D+2019.
Fixes #3995
closed 2023-09-29 by raphaelcoeffic