EdgeTX/edgetx #4002: EdgeTX 2.9.0 does not boot on Jumper T-Lite with F4 MCU

AndrewBCN, 2023-09-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

This is a repeat of issue #3947, which was closed because the bug was supposedly fixed after 2.9.0RC2. Unfortunately it seems the bug is still there, as 2.9.0 (final) does not boot when flashed to a Jumper T-Lite with F4 MCU.
2.8.5 boots fine.

### Expected Behavior

EdgeTX 2.9.0 is expected to boot fine when flashed on Jumper T-Lite with F4 MCU.

### Steps To Reproduce

Flash EdgeTX 2.9.0 using Buddy to Jumper T-Lite with F4 MCU.
Press Power button to turn on radio.
Nothing happens, screen remains dark.
Re-flash 2.8.5.
Pressing Power button turns on radio normally.

### Version

2.9.0

### Transmitter

Jumper T-Lite

### OS Version

Windows 11 with Chrome

### Anything else?

Repeat of issue #3947, which was supposedly fixed. Unfortunately, the bug is till there.

pfeerick, 2023-09-04:
Flash the T-Lite firmware, not the T-Lite (F4). Jumper has the F4 MCU configured to run in a compatibility mode so it won't run F4 firmware.

closed 2023-09-04 by pfeerick

AndrewBCN, 2023-09-04:
Thanks, the T-Lite 2.9.0 firmware does indeed boot fine on my Jumper T-Lite V2 and seems to work without issues.
May I suggest adding a mention of this issue in the "Release Notes" for 2.9.0?
And as usual, thank you very much for your fine work.

3djc, 2023-09-04:
We should remove/hive TLite F4 option altogether, as this type of radio hasn't been produce yet (not fully sure it will ever be)

AndrewBCN, 2023-09-04:
It seems Jumper decided to replace the STM32F205 in the T-Lite V1 with a cheaper CKS32F407 in the T-Lite V2, perhaps because of availability and cost issues during the "chip shortage crisis" in 2021/2022.
I agree with you, I don't think they'll ever go back to an STM32F407 MCU for a hypothetical T-Lite V3, why would they since the CKS32F407 clone is cheaper and works fine with EdgeTX?
Just a note that I found a datasheet (in Chinese) for the CKS32F407 MCU, I am attaching it to this comment.
[CKS32F407VGT6.pdf](https://github.com/EdgeTX/edgetx/files/12512700/CKS32F407VGT6.pdf)

pfeerick, 2023-09-04:
>as this type of radio hasn't been produced yet

Perfect, that was what I was going to ask next, as if it's not in production yet/ever, it can get removed from Buddy for now. I'll pull it it from 2.9 and main lists, but I have a feeling Buddy won't pick it up for v2.9.0, only the next tagged release, so will need to add a note in the release notes.