EdgeTX/edgetx #4006: Radiomaster Pocket is supported but it's firmware is not built

Gruppio, 2023-09-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Build system

### Current Behavior

Radiomaster pocket is supported [PR](https://github.com/EdgeTX/edgetx/pull/3892) but it is not listed in the supported radios and it's firmware it's not built.

### Expected Behavior

Add it

### Steps To Reproduce

None

### Version

2.9.0

### Transmitter

Other (Please specify below)

philmoz, 2023-09-05:
It will be in 2.10.
No new radios were added to 2.9 as far as I know.

pfeerick, 2023-09-05:
@philmoz is correct. The only new radio that was added in 2.9 was the RM Boxer, as it was added very early on in the 2.9 cycle. The Jumper T20, TPro V2, RM Pocket and any other radios that come along in time will be supported in 2.10, and with the initial support present in the nightlies for 2.10.

closed 2023-09-05 by pfeerick