EdgeTX/edgetx #4007: Allow Global Variables as source for Inputs & Mixers

DocTrucker, 2023-09-05:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am trying to use a continuous global variable (GV1 in this case, that is set using special functions) to feed an output channel. Currently the only ways I have come across to do this are obscure and not intuitive.

### Describe the solution you'd like

Allow the use of global variables such as GV1 as sources for inputs and mixers. This would make it more intuitive and consistent with the normal work flow.

### Describe alternatives you've considered

Tested: Use a mixer with MAX source and weighted to GV1.

Alternative recommended but not tried: Create a new input, pick and source. Set weight to 0% and offset to GV1.

philmoz, 2024-08-03:
Global Variables can be selected as the Source for an Input.

Unsure why they cannot be selected for mixer lines.

DocTrucker, 2024-08-04:
I think the source for input has been soulrted since I raised the report last year.

closed 2024-08-07 by pfeerick

DocTrucker, 2024-08-08:
Great! I did notice global variables being allowed as inputs recently, but +/- 100% global was only resulting in +/- 9.8% as an input. I'll test 2.10.3 later and file a new bug if still not working right.

pfeerick, 2024-08-08:
Have a look at #5403 before you do ;) This is actually expected behaviour, as the mixer is expecting values between -1024 and 1024, but we need to do something to allow for use as a mixer value.