EdgeTX/edgetx #4018: TX 16 S EdgeTX 2.9 How to remove the vibrator when switching on the transmitter?

LS-67, 2023-09-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Even choosing in the radio settings "None" for the vibrate, it still works when turning on the transmitter...

### Describe the solution you'd like

I would like to have to choose in the radio parameters to complete disable the vibrator (even for turning on (or off) te transmitter)

MRC3742, 2023-09-07:
I thought I might be the only one that finds haptic at startup very annoying?
Overall I find haptic annoying for everything except Alarm Mode only.
I currently self build and comment out this line to eliminate the startup buzz.
https://github.com/EdgeTX/edgetx/blob/d3ed7c12e63d12cd3925d233a6fedaf146e104a1/radio/src/opentx.cpp#L1485

Please make the haptic at startup a user configurable option!!!

Thank You - Rich

pfeerick, 2023-09-07:
Thanks for opening the issue - without this it's hard to gauge how many people do or don't want a particular behaviour ;) Currently, it's not possible, as the vibration happens before the SD card is even read. However, it looks like it should be possible to simply move the vibration call down a bit further, until after the initial unvalidated read of the configuration values is done, and therefore do the vibration unless configured not to do so.

3djc, 2023-09-07:
Moving it later would have a drawback, because haptic is when you can release the power button without interrupting the start. So it's a signal saying you can stop pushing

philmoz, 2023-09-07:
On B&W radios the radio.yml file is loaded before the haptic is played (call to storageReadRadioSettings in opentxInit) so it is possible to use the radio haptic setting to control the haptic play. I've tested this on X7S.

The call to storageReadRadioSettings could be done for all radios - I tried this on the TX16S and EL18 and it does not appear to cause any issues. Of course it will mean a slightly longer startup time.

I did also note that storageReadRadioSettings is always called - even for an EM startup. Not sure this should happen as storageReadRadioSettings calls sdInit.

pfeerick, 2023-09-07:
> Moving it later would have a drawback, because haptic is when you can release the power button without interrupting the start. So it's a signal saying you can stop pushing

I was thinking would need to see how much of a time difference it would be ... i.e. I doubt the user is going to notice a 10ms difference, but you are right, since it's part of the animation sequence on B&W the check would need to be there... which is entirely possible as Phil indicated as the unvalidated load of settings is done early enough on B&W. 

> On B&W radios the radio.yml file is loaded before the haptic is played (call to storageReadRadioSettings in opentxInit) so it is possible to use the radio haptic setting to control the haptic play. I've tested this on X7S.

The issue here is more colorlcd centric, as B&W radios with pushbuttons (and startup animation enabled) have always had the haptic feedback, and it's only colorlcd that has changed recently, hence users not used to the behaviour. Maybe we instead always do the 
```cpp
  bool radioSettingsValid = storageReadRadioSettings(false);
  (void)radioSettingsValid;
```
and then for both `runStartupAnimation` and the colorlcd `haptic.play(15, 3, PLAY_NOW);` instead only play the haptic if enabled? Could possibly also then also skip `storageReadRadioSettings()` if `UNEXPECTED_SHUTDOWN()`?

LS-67, 2023-09-10:
Or, could we simply consider reducing the strength of the vibrator when the TX is switched on ? (at the same strength as when the TX is currently turned off)

LS-67, 2023-09-30:
In which version of ETX will this modification be implemented?

pfeerick, 2023-09-30:
The PR that will add options for it will be in 2.10, but I may l do a custom build that removes power-on haptic for colorlcd, which would be as part of the 2.9.2 custom and language builds.

LS-67, 2023-09-30:
Thank's a lot  ;)

PR #4027 (2023-11-05): chore(radio): Cleanup startup code.
Fixes #4018 

Summary of changes:
- Remove SPLASH build option
- Move splash handling code to gui files/folders
- Rename opentxXXX functions to edgeTxXXX
- Fix throttle warning startup timing (haptic played before splash screen finished on color LCD)
- Use radio setting to control startup/shutdown haptic
PR #4285 (2023-11-06): fix(color): No startup/shutdown haptic if haptic mode set to quiet

Resolves #4018 for 2.9

Ports the following from #4027:
- Link startup and shutdown haptic triggers to state of hapticMode setting
- Include throttle LED fix
- Load settings on boot unless EM startup so haptic state can be loaded on boot
closed 2023-11-05 by pfeerick