EdgeTX/edgetx #4021: v2.9.0 DSM telemetry ESC ECUR value reporting 10x higher than v2.8.5

silicongarage, 2023-09-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

ESC:  Eflite EFLA1030E (smart)
RX:    AR8360T

v2.9.0 telemetry reports ECUR 10x higher than v2.8.5

With my ESC and v2.8.5, the ECUR RATIO is set to 295 (about +16%) to correctly report current and is within expected tolerance for low end ESC current measuring performance. This RATIO has been validated by matching ECUR Consumption with actual LiPo charger capacity.
To get v2.9.0 to report correct ECUR, RATIO has to be set to ~30 for a scaling factor of 30/255 to get a current that looks correct.

### Expected Behavior

ECUR should report same value as in v2.8.5

### Steps To Reproduce

1. Display ECUR using same RATIO factor as from v2.8.5

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

frankiearzu, 2023-09-08:
There was a bug on previous version..  with some of this ESC measurements... I did the changes to match the doumentation and validating by comparing the values to what a Genuine Spektrum TX shows in the screens.   Let me do some tests again and will post some screenshots.

frankiearzu, 2023-09-08:
@silicongarage (we are in the same City)...  After reading more the bug report... what is your understanding of what the "ECUR" is??    i think that you are thinking that is the battery capacity..  when you mention: "validated by matching ECUR Consumption with actual LiPo charger capacity"..    the ECUR is real time current, not accumulated consumption.    Anyway, will do a few test this afternon.

frankiearzu, 2023-09-08:
Validated the Data against the documentation, as well as comparing it to the readings of a genuine Spektrum TX at 3 different settings of Throttle (no prop, so no load).  No Ratio.. pure raw data.  Everything seems to match.

About 12-15%, 50%, and 100%.     Different times (voltage can be a bit different). and my stick calibratio is like 3% of from the Spektrum TX one.

[image]
[image]
[image]

frankiearzu, 2023-09-08:
Here is another one with a Blade 230s Helicopter, who has an Smart Bat too..   They too are close..   About same readings with an Spektrum TX.

[image]

silicongarage, 2023-09-09:
Hi Frankie!

Sorry if I caused confusion about the  ECUR sensor value validation. What I meant to say was I also created another sensor "Etot" that is calculated using "Consumpt" of ECUR.   I then tweaked ECUR RATIO slightly from the default unit precision of 255 until Etot matches the value of MaH I get when charging the pack back up.  For v2.8.5 the RATIO (for my ESC) ended up around 295 or about +16% of the raw default ECUR.  This seems entirely reasonable as the RATIO for ECUR is really close to the default raw ECUR value.

I tried what you suggested and deleted ECUR and let it be rediscovered.  What I just found doing this was that with RATIO = 0,  the reported ECUR is really close to my un-tweaked value of ECUR value in v2.8.5.  However when I set RATIO=295, ECUR is now about 10x too large when it should have only increased ~16%.  So it appears the problem is with trying to scale ECUR with the RATIO value.

I have based all my scaling calculations for RATIO from looking at the code in "telemetry_sensors.cpp" lines 699-710.  From this code the unit RATIO value should be 255.  As an aside, the value of '122' on line 709 should probably be 127 or 128 to correctly deal with the truncated integer division by 255, but the error from 122 is probably insignificant. 

Haha, I need to adjust the RPM sensor - 752K seems a little excessive!

[image]

[image]

[image]
  
RATIO=0
[image]

RATIO=295
[image]

silicongarage, 2023-09-10:
Here is what my NX10 reports
[image]

3djc, 2023-09-10:
So working with ratio=0, which is what is expected, no ?

silicongarage, 2023-09-10:
With RATIO=0, v2.9.0 reports close to NX10.   However when trying to fine tune ECUR value to account for ESC ADC measurement (gain) error by adjusting RATIO the value is off by around 10x.  

RATIO=0 is default scaling of 1.  RATIO=255 should also be default scaling of 1.  However RATIO=255 reports 10x the value of RATIO=0 in v2.9.0.

In my use case I have tuned RATIO=295 which should be a scaling factor of 295/255=1.16 to give a true ESC correct ECUR. 
In v2.8.5 RATIO=295 scales ECUR by 1.16,  in v2.9.0 it scale ECUR by 11.6

Correctly tuned ECUR (and hence ETOT Consumption) is important as I use it to determine remaining battery flight capacity on my e-gliders.

frankiearzu, 2023-09-12:
Took a look at the code in  "convertTelemetryValue".. depending on the precision, they multiply by presision before the conversion,  but after the conversion, they divide by precision. Totally make sense, since the number is stored as an integer, the precision is just for display.

[image]

I think the "getValue" has to do something similar... divide after applying the ratio.

[image]

That is where the 10x comes when precision=2.   Before this bug was not doing the 10x, because it was in "mA", and precision was 0.

i think a /10 (div) needs to be added before applying the offset when prec=2...   Or do we need the first multiplication at all??Will play with some numbers... but this affects every telemetry protocol, not just Spektrum.

frankiearzu, 2023-09-12:
Looked at the code in OpenTX... this was changed.. instead of the "ratio" there is a "multiplies" and "divisor" fields.

[image]

frankiearzu, 2023-09-20:
@silicongarage 
Can you tried this firmware.   Is compiled from the v2.9 branch, with the change
to eliminate the code doing prec checking.  Tried the ECUR as well as other sensors ans seems ok.

https://github.com/frankiearzu/DSMTools/blob/main/EdgeTx_Firmware/tx16s-edgetx-v2.9.0-ratio-txt.bin

[image]

silicongarage, 2023-09-26:
Just verified that your patch gives correct value now using RATIO scaling. Tested with  PREC={0,1,2}

PR #4083 (2024-01-07): fix: Properly scale PREC2 sensor values using "ratio"
Fixes #4021

History:
During the transition from 2.8.5 to 2.9.0, there was several fixes to DSM sensor values.   The values itself are correct, but it exposes a poblem when using "ratio" to adjust the value of a sensor.

The "ECUR" (ESC Current) sensor change from been in "mAh" (prec=1) to "A" (prec=2).  The change of prec exposed the scaling problem with ratio, The value in 2.8.5 got 10x bigger in 2.9.0 when using ratio.

Summary of changes:
* Simply commenting out code that doing the x10 when prec==2.
* The other problem was when prec==0, [... cut]
closed 2024-01-07 by pfeerick