EdgeTX/edgetx #4023: SD card error when logging is active and USB connected/disconnect

mha1, 2023-09-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

- Activate SD card logging
- Plug in USB connection to PC while logging is active
- select USB storage
- disconnect USB conection
- SD card error message pops up

### Expected Behavior

No error message

### Steps To Reproduce

see section current behavior, also present in 2.9 an main

### Version

2.8.5

### Transmitter

Flysky NV14

### Operating System (OS)

Windows

### Anything else?

my bad, missed a use case

pfeerick, 2023-09-07:
lol... ouch!

mha1, 2023-09-07:
I hear you but at least you gotta give me some credit for finding and fixing my own oopsies

pfeerick, 2023-09-07:
I'll give you extra credit if you don't make them in the first place. 😛 Actually, on second thoughts...  keep making them... else I need to not make any either! 😮

PR #4024 (2023-09-24): fix: Deactivate logs and close log file if USB is connected
Fixes #4023
closed 2023-09-24 by pfeerick