EdgeTX/edgetx #4028: Jittering in Joystikcs.  There is movement in joysticks

Talha909, 2023-09-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I am using the edge TX xlites 2.10.o GIT 2F3C1043 . There is no jittering in joystiks in that version. After that 2.9 rleased I update the trasmitter to the 2.9. Now the jittering is back. I also tried the ADC fillerter on and off but there is no change in jittering. But in 2.10 there is 0% jittering. Whats the reason ?

### Expected Behavior

Jittering will be gone in 2.9 

### Steps To Reproduce

I did not touch the joystick but there is some movement in all the sensors. 

### Version

2.9.0

### Transmitter

FrSky X9 Lite / Lite S

### OS Version

Big

### Anything else?

[](url)
[image]

3djc, 2023-09-10:
Previously, the joystick was displayed as 0.0 format, which always had some level of jitter, even on perfectly fine joystick.

Display was changed to 0 (decimal was dropped) , so you are comparing two different things

Talha909, 2023-09-11:
Yes I am comparing 2.9 and 2.10 nightly. There is no jittering in 2.10 nightly but its comes in 2.9. upper video is for 2.10 nightly and 2nd video is with 2.9 final.

3djc, 2023-09-11:
Look at the numbers: 2.9 does have decimal values, when 2.10 doesn't , the jitter you see comes from the decimal value

closed 2024-04-08 by pfeerick