EdgeTX/edgetx #4030: EMERGENCY MODE WHEN ASKING EXTERNAL ANTENNA

ds7060, 2023-09-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If you have a model with the option "ASK" for selecting internal or external antenna on startup, and you have any switch in a different position than the model default positions, system crash when you put the switches to the required position and goes to EMERGENCY MODE without asking for the antenna type.

### Expected Behavior

It should ask for the antenna as normal startup.

### Steps To Reproduce

Setup a model that ASK for use internal or external antenna. ( I am using X12S with IXJT )
1. Put a switch to a position that is not normal for your model.
2. Start radio
3. System ask you to correct switch position and when you do it it crash
(if the switch is in the correct position when starting, system ask you to select the antenna, internal or external and it starts normally and works without issues for hours)

### Version

2.9.0

### Transmitter

FrSky X12

### Operating System (OS)

Windows

### OS Version

W10 PRO 22H2

raphaelcoeffic, 2023-09-11:
Could you please upload the `radio.yml` and model file (`modelXX.yml`) in question? At best together in a ZIP file to avoid confusing GH ;-) I'm having issues reproducing that here on simulation.

pfeerick, 2023-12-24:
@ds7060 Can you retry this with a nightly, as I believe there was a fix around this popup. Backup your RADIO and MODELS folders first so you can roll back afterwards.

closed 2024-05-06 by github-actions[bot]