EdgeTX/edgetx #4031: Can't set Timer1 ON

ask6483, 2023-09-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I can't set Timer 1 setting to 'ON' in some of the models.
'ON' setting seems missing on the list. At single scroll event setting is not jumping to the next item just stays at 'OFF'. You have to scroll quickly to jump to the last item - 'THt' but scrolling back by one there is no 'ON' item - after 'Strt' there is 'OFF'. 
Other timers are working correctly, also tried to modify yaml file by hand but this doesn't fix the problem.

### Expected Behavior

Timer 1 should be turned 'ON'.

### Steps To Reproduce

1. Enter model setting
2. Scroll to Timer 1
3. Try to set in 'ON' - there is no option to scroll to other value than 'OFF' without quick jump to the last item.
4. Scroll back from 'Tht' to reach 'ON' setting - there is no 'ON' after 'Strt'.
5. We are back to 'OFF'

### Version

2.9.0

### Transmitter

FrSky X9D+2019

### Operating System (OS)

Linux

### OS Version

Debian 12 testing

ask6483, 2023-09-10:
Link to video with presentation of the bug:
https://nextcloud.andrzejskiba.com/index.php/s/HLSZLdjgJ49wYL4

philmoz, 2023-09-11:
If it only happens on some models then there must be something specific to the affected models - the code for each timer is identical so I suspect some external influence (like a Lua script for example).

Can you upload the affected modelXX.yml file and the radio.yml file from the radio (you will need to rename them to upload).

closed 2023-09-11 by 3djc