EdgeTX/edgetx #4041: Global Functions not available in System Menu if not enabled for current model

rhmaddox, 2023-09-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When Global Functions are disabled in a model.  The System Menu does not show the Global Functions.

### Expected Behavior

System Menu should not be effected by the Current Model Settings.

### Steps To Reproduce

Turn off global functions on a model.
Select System Menu
Global Functions are not available.

Turn on global functions on a model.
Select System Menu
Global Functions are available.

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

ParkerEde, 2023-09-14:
yes exactly. That's where this ticket came from. 
https://github.com/EdgeTX/edgetx/issues/3980

At the beginning, however, I didn't know what the problem was.
Since 2.9.0, the visibility of the menu is also mixed with the use of the global functions.

raphaelcoeffic, 2023-09-14:
See here: https://edgetx.gitbook.io/edgetx-user-manual/edgetx-user-manual/user-manual-for-color-screen-radios/radio-settings/radio-setup/additional-radio-settings#enabled-features

And here: https://edgetx.gitbook.io/edgetx-user-manual/edgetx-user-manual/user-manual-for-color-screen-radios/model-settings/model-setup/enabled-features

rhmaddox, 2023-09-15:
Ok I don't agree with the manual.   Why would a model menu configuration be
reflected in the System menu?   System menu means, to me , over all radio
configuration.  Not a model specific setting.

But i get it,  Just my thoughts

Thanks
Rick