EdgeTX/edgetx #4053: Build FW in Browser with gitpod actualy not possible

3DRacer, 2023-09-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Build system

### Current Behavior

[image]

Oh, no! Something went wrong!

supervisor run error with unexpected exit code: IDE failed to start: exit status 1

### Expected Behavior

site starts without error

### Steps To Reproduce

try to open https://github.com/EdgeTX/edgetx/wiki/Building-radio-firmware-in-a-webbrowser-with-Gitpod

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

11

### Anything else?

[image]

koaxheli, 2023-09-17:
I have the same error since the day before yesterday.

Oh, no! Something went wrong!
supervisor run error with unexpected exit code: IDE failed to start: exit status 1

 
edgetx-edgetx-ltm7g3kmul2

https://github.com/EdgeTX/edgetx/tree/main

koaxheli, 2023-09-20:
Error message is not yet removed.
Still exists when compiling under Gitpod.

[image]

raphaelcoeffic, 2023-09-20:
Guys, please test using #4065.

3DRacer, 2023-09-20:
> Guys, please test using #4065.

compiling FW for the RM TX16S and download ..... success

koaxheli, 2023-09-20:
Tested: 
EdgeTX version 2.9 is OK. Tested with #4065

But the latest version 2.10.0 still kept the bug.

The call on Gitpod is the following, which has always worked so far.

https://gitpod.io/#https://github.com/EdgeTX/edgetx/tree/main

koaxheli, 2023-09-23:
I have a request, can you still provide an update for the main development branch 

https://gitpod.io/#https://github.com/edgetx/edgetx/tree/main

as you did for

https://gitpod.io/#https://github.com/edgetx/edgetx/tree/2.9-gitpod-workspace

As for the v2.10.0 on Github also runs into the same error.

Thanks

pfeerick, 2023-09-23:
Of course. #4065 need to be merged first ;)

closed 2024-04-08 by pfeerick