EdgeTX/edgetx #4059: Jumper T-Lite speaker noise after updating from 2.8.5 to 2.9

mar-hok, 2023-09-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After updating my Jumper T-Lite v1 from 2.8.5 to 2.9 with Companion 2.9.0 Linux, my T-Lite started emitting a buzzing noise from the speaker. This did not occur before updating. I have 2 such radios, and repeated the update procedure carefully to the another after finding that odd behaviour with the first TX update. The same buzz appeared in both cases exactly the same manner after update. The buzz changes tone or diminish almost off if I touch the antenna when the tx is on.  This bug is rather annoying, rendering the voice feedback almost useless. 

### Expected Behavior

No buzzing or other noise from the speaker.

### Steps To Reproduce

1 - Start with 2.8.5, flash to 2.9 using latest Companion, proceed acc. to instructions, exit from the bootloader mode and restart TX normally as starting to fly a plane.
2 - After the occasional throttle warning if throttle is not at min, when the led turns from red to green, a buzzing noise starts to emit from the speaker, not very load but enough to annoy. 

The buzzing sound changes if you touch the antenna. 

Since this buzzing did not happen when using 2.8.5, this can not be purely a hardware problem.

### Version

2.9.0

### Transmitter

Jumper T-Lite

### Operating System (OS)

Linux

### OS Version

Ubuntu 22.04 LTS

mar-hok, 2023-09-19:
Not a bug but a feature ( audio mute )

TonyNJFPV, 2023-09-28:
Yes. After update on my iFlight Commando 8 from 2.8.0 to 2.9.0 - I now have low volume beeps any time I get to page 1/7. Like there's a running error in the background.

pfeerick, 2023-09-28:
Turn on the "Audio mute" in RADIO -> HARDWARE ... it is a new option in 2.9 and is off by default, as muting the audio amp when not in use adds a delay for rapid playing audio such as variometer.

TonyNJFPV, 2023-09-29:
Thanks, apologies. I'm just dense. Much appreciated.

pfeerick, 2023-09-29:
Nah, can't been on top of all the new stuff and it's implications...

TonyNJFPV, 2023-09-29:
> Nah, can't been on top of all the new stuff and it's implications...

One dumb question. How do I save so I don't need to check the box every time I turn on the TX?

3djc, 2023-09-29:
It should stick automatically, I'll do some testing tomorrow

TonyNJFPV, 2023-09-29:
Thanks, yeah... not sticking. I'm on an iFlight Commando 8 (do not like it!)

pfeerick, 2023-09-30:
It should, but it isn't ... iFlight Command8 on 2.9.1 is not retaining the setting across reboots, Jumper TLite is.

TonyNJFPV, 2023-09-30:
I won't downgrade as nothing else seems bad. Just wanted to be sure I didn't F up. Back in this hobby after 10 yrs, so getting used all this cool new stuff.

TonyNJFPV, 2023-09-30:
That fix is above my paygrade, rofl. Also, after the upgrade to 2.9.1, I no long have the serial port option. Only joystick and USB storage. Strange.

pfeerick, 2023-10-01:
I'm not sure VCP was actually valid for that radio, as IIRC passthrough ELRS flashing isn't supported (which is the normal reason for it to be shown as an option, outside of debug builds).

TonyNJFPV, 2023-10-01:
Well, if this is the case I have no way at all to update ELRS to 3.x.x as WiFi fails continually. This pos is going back. Bad decision on my part. Thanks for the info.

pfeerick, 2023-10-01:
Undo the two screws holding the module into the transmitter and pull it
out. the module should have its own USB port at the bottom for UART
flashing in ELRS Configurator.

TonyNJFPV, 2023-10-03:
> Undo the two screws holding the module into the transmitter and pull it out. the module should have its own USB port at the bottom for UART flashing in ELRS Configurator.
> […](#)

That worked, thanks a ton! It somehow disturbed the mixes in my single model, but other than that binding phrase works and all seems ok. Much appreciated.

pfeerick, 2023-10-03:
Fantastic. The most recent 2.9.1 may have unearthed mixes that were
actually there but hidden due to a UI bug.

The Command8 has a few little idiosyncrasies, but is quite a nice radio to
hold and use. Feels very solid, and switches feel good on it.

TonyNJFPV, 2023-10-03:
It does feel good, but mine came with a yaw gimbal that won't center (only 2 adjustment screws that don't help) and though it sounds/looks trivial I can feel it on takeoff and do have a tendency to yaw left due to the offset. 

I don't fully understand the mix page but mine looks like this (see below), and works again. But it isn't how other people's settings look around the web. I have to spend a few hours really learning EdgeTX. 

[image]

[image]

pfeerick, 2023-10-03:
Yes, that is probably the only thing that leaves room for improvement... the gimbals. They are good and bad IMO 😆 I think with the Boxer you will find things at lot similar, and also some stuff has been implemented a lot better there (i.e. gimbals are much nicer, more intuitive radio navigation controls, passthrough for ELRS recovery, etc). 

On your inputs screen, the name missing for the Rudder control makes me think the name for the rudder stick is messed up, hence the `---`. You should be able to go into SYS -> HARDWARE,  go to the rudder stick line (which should say `---` like the other lines), and as strange as it sounds, simple press Enter to select it, and make sure each character is blank (blank, not a dash). When you go back to the inputs page, it should have reverted to saying Rud.