EdgeTX/edgetx #4062: GPS telemetry  variable names (lat/lon both named "GPS")

f2knpw, 2023-09-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

Using OxS on RP2040 and connecting a GPS to it, I got a working module to send telemetry via my FS-IA6B receiver FlSky2A protocol.

Telemetry is discovered but latitude and longitude have the saml "GPS" name. (which is not useful in lua scripts).

Mstrens (author of OxS on RP2040) told me that this :

"_In flysky (Ibus) protocol, oXs does not send the name of the field but a code.
The codes are defined like this:
IBUS_SENSOR_TYPE_GPS_LAT = 0x80, //4bytes signed WGS84 in degrees * 1E7
IBUS_SENSOR_TYPE_GPS_LON = 0x81, //4bytes signed WGS84 in degrees * 1E7

So oXs uses 2 different codes for long and lat.
I expect that it is EdgeTX that converts the code into a free text.
I suggest that you ask on EdgeTx forum because I expect the issue is at this level._"

For clarity hereis the picture showing the "bug"

[image]

I had to change the names:
[image]

This is of bourse not a critical bug. But every time I "re discover" telemetry the names are reset to "GPS" !

### Expected Behavior

change "GPS" by "Lat" and "Lon" (or "Glat" "Glon")

Thanks

### Steps To Reproduce

discover a telemetry with GPS inside.

### Version

2.9.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

full explaination of the project here : https://hackaday.io/project/192752-variometer-gps-tracker-remoteid-for-rc-glider/details

(bug explained after the middle of the details page)

[image]

mha1, 2023-09-19:
Just a side note: I think the google earth feature if Companion's log viewer will only work with lat/lon combined as sensor GPS. The telemetry screen is also capable of showing the combined lat/lon with sensor GPS.

frankiearzu, 2023-09-20:
Why not use the GPS for the combined coordinate (Lat + Long).
I did a fix related to 0xS for the DSM protocol, and the GPS sensor can hold both values:

[image]

philmoz, 2023-09-20:
Some telemetry protocols send a single value with both Lat and Lon. Others have separate values for Lat and Lon.

frankiearzu, 2023-09-20:
@philmoz DSM send it separate too, but when processing the value, you can tell if you are setting the Lat or long portion  when populating the GPS sensor, At least in the Spektrum Telemetry. let me see how is is done for FrSky

mha1, 2023-09-20:
use `setTelemetryValue(TelemetryProtocol protocol, uint16_t id, uint8_t subId, uint8_t instance, int32_t value, uint32_t unit, uint32_t prec)` and call it for 
- latitude: with value = latitude, unit = UNIT_GPS_LATITUDE, prec = 0
- longitude: with value = longitude, unit = UNIT_GPS_LONGITUDE, prec = 0

frankiearzu, 2023-09-20:
just read  the code of flysky_ibus.cpp   This is how the GPS can be populated with the combined LAT/LON.  Don't have a GPS sensor, so don't know if the value needs any conversion.

1. From what i can see in the code, there is a part where they parse telemetry message with combined LAT/LON/ALT and split it into single sensors.. Probably the protocol support both,  comming as individuals or as a combined packed.

2. To ensure that the individuals stays there (for testing, can be removed later). Added a new virtual sensor that is the combined GPS. this sensor needs to be of type UNIT_GPS to be display and do distance math correctly.

3. in the "processFlySkySensor(const uint8_t * packet, uint8_t type)" added special code to handle when latitude or logitude comes individually, populate the combined GPS lat or lon portions.

Complete file: (just renamed to .cpp)
[flysky_ibus.txt](https://github.com/EdgeTX/edgetx/files/12669255/flysky_ibus.txt)

[image]

frankiearzu, 2023-09-21:
@f2knpw:  Can you try this firmware.. it only going to report one GPS sensor, but it will have both Latitude/Longitude.
The default display format for the radio is in DMS (Deg-Min-Sec), but you can change it in System->"RADIO Setup"->GPS to be NMEA (Deg + Decimal Minutes) that will be closer to the value sent.   Currently i am dividing the value by 10 (others do that), but might need a different division to get the correct value.  

Firmware for TX16S: Taking from dev branch for 2.9.x (2.9.0 + a few later changes... probably just use it in the ground until more testing).
https://github.com/frankiearzu/DSMTools/blob/main/EdgeTx_Firmware/tx16s-edgetx-v2.9.0-ibus-gps.bin

f2knpw, 2023-09-21:
Thank you for this reply.

Values are indeed sent multiplied by 10000000. So to get decimal values you should have to divide by 1E7.

I will test the firmware, but, honestly I don't understand why lat/lon values are merged into a single field. I will have to split them into latitude and longitude to be able to use them !
A GPS value is not only latitude and longitude... but also altitude, heading, speed... and so on. They could have been merged into an even bigger "GPS" !

But that's not a big deal, I can cope with it !

mha1, 2023-09-21:
how do you use latitude and longitude then?

f2knpw, 2023-09-21:
I do use latitude and longitude for a GPS tracker and a "remoteID" beacon. So I need a way to get these two values easily.
Currently I wrote a lua script doing this 
`local function init()
  -- Called once when the script is loaded
	mlat =  getFieldInfo("lat").id    -- here we get the numerical id of the filed "lat"
	mlon =  getFieldInfo("lon").id    
	mGSpd =  getFieldInfo("GSpd").id    
	mHdg =  getFieldInfo("Hdg").id 
	mGAlt =  getFieldInfo("GAlt").id 	
end

local function run()
  -- Called periodically while the Special Function switch is on
--{"OTA":0, "lat":435128245, "lon":14970095, "GSpd":0, "Hdg":193.33999633789, "GAlt":183.60000610352}
	serialSetPower(1, 1) -- setPower port AUX2 on --> serialSetPower(port_nr, value)
	serialWrite('{"OTA": 0, "lat":'..getValue(mlat)..', "lon":'..getValue(mlon)..', "GSpd":'..getValue(mGSpd)..', "Hdg":'..getValue(mHdg)..', "GAlt":'..getValue(mGAlt)..'}\n')
end

local function background()
  -- Called periodically while the Special Function switch is off
end

return { run=run, background=background, init=init }`

it is for this project : [https://hackaday.io/project/192752-variometer-gps-tracker-remoteid-for-rc-glider](url)

mha1, 2023-09-21:
```
		local fi = getFieldInfo(sensor)
		
		local name
		local value
		local unit
		local found
		
		if(fi == nil) then
			found = false
		  name  = sensor
		else	
		  found = true
		  name  = fi.name
			value = getValue(fi.id)
			unit  = fi.unit

			-- special treatment for GPS sensor (lat/long)
			if name == "GPS" then
			  if type(value) == "table" then)
					value = value.lat.." "..value.lon
				else
					value = "-no data available -"
				end
			end 
		end
```

value.lat and value.lon hold latitude and longitude for you to process individually. In my case I just concatenate them for display on a single line.

All of OTX's and ETX's processing of GPS data is built on "GPS" and "GPS" holding both lat and lon, e.g. logging, log viewing, calculated sensor "distance", etc.

f2knpw, 2023-09-21:
Thanks for this code.
so value.lat and value.lon keep track of latitude and longitude.
Is it also the case for altitude ?  To compute distance we would need lat, lon and altitude ?

Sorry for all these questions I am a complete noob in lua and discovered EdgeTx last week (and I love it)

To use this code (to get all GPS values from telemetry) what should I do ?

mha1, 2023-09-21:
Altitude is not part of the GPS table. You fetch altitude from your preferred altitude sensor, mostly  "Alt"

3djc, 2023-09-21:
> Thanks for this code. so value.lat and value.lon keep track of latitude and longitude. Is it also the case for altitude ? To compute distance we would need lat, lon and altitude ?
>
> Sorry for all [...]

You have built in code to compute distances (with or without alt): create a calculated sensor of type distance

f2knpw, 2023-09-21:
> Altitude is not part of the GPS table. You fetch altitude from your preferred altitude sensor, mostly "Alt"

Ok that sounds easy !

But with my lua script (above) how do I get the lat and lon ?

I did this 👍 
local function init()
-- Called once when the script is loaded
mlat = getFieldInfo("lat").id -- here we get the numerical id of the filed "lat"
mlon = getFieldInfo("lon").id
mGSpd = getFieldInfo("GSpd").id
mHdg = getFieldInfo("Hdg").id
mGAlt = getFieldInfo("GAlt").id
end

should I replace "lat" and "lon" by "GPS.lat" and "GPS.lon" ? or something more complex ?

f2knpw, 2023-09-21:
@mah1 --> thanks so much. That's what I was looking for !

mha1, 2023-09-21:
@f2knpw https://luadoc.edgetx.org/part_iii_-_opentx_lua_api_reference/general-functions-less-than-greater-than-luadoc-begin-general/getvalue

mha1, 2023-09-21:
everything was and is working as specified. If with "modified firmware" you are referring to @philmoz's PR: this PR is not necessary if you ask me. So is your issue which should be closed.

frankiearzu, 2023-09-21:
@f2knpw: This firmware puts the latitude/longitude into a single sensor.. also will do the proproper conversion to get the right value..  
Will be good to get it fixed, since it was not working as expected for the ibus protocol. The value of the sensor might need some more changes to get it right,   

In one of your comments "Values are indeed sent multiplied by 10000000. So to get decimal values you should have to divide by 1E7."  

So that means that the proper conversion should be "value = raw_value / 487"   (487=0x1E7).
I can add that, but try the current one that is only dividing by 10.
Don't have the hardware to test it, so we need your help!

Firmware:  Should be a single sensor with Lat/Lon together.
https://github.com/frankiearzu/DSMTools/blob/main/EdgeTx_Firmware/tx16s-edgetx-v2.9.0-ibus-gps.bin

f2knpw, 2023-09-21:
Hi 

should be divided by 10000000 decimal. (not 0x1E7)

I will test it !

f2knpw, 2023-09-21:
Meanwhile I have tested @mha1 piece of code with current firmware.

I do use this script to output lat lon to serial port Aux2

local function init()
  -- Called once when the script is loaded
	mGPS =  getFieldInfo("GPS").id    -- here we get the numerical id of the field "GPS"   
	mGSpd =  getFieldInfo("GSpd").id    
	mHdg =  getFieldInfo("Hdg").id 
	mGAlt =  getFieldInfo("GAlt").id 	
end

local function run()
  -- Called periodically while the Special Function switch is on
--{"OTA":0, "lat":435128245, "lon":14970095, "GSpd":0, "Hdg":193.33999633789, "GAlt":183.60000610352}
	serialSetPower(1, 1) -- setPower port AUX2 on --> serialSetPower(port_nr, value)
	serialWrite('{"OTA": 0, "lat":'..getValue(mGPS).lat..', "lon":'..getValue(mGPS).lon..', "GSpd":'..getValue(mGSpd)..', "Hdg":'..getValue(mHdg)..', "GAlt":'..getValue(mGAlt)..'}\n')
end

local function background()
  -- Called periodically while the Special Function switch is off
end

return { run=run, background=background, init=init }

But it doesn't work.

While this one works fine (after renaming GPS fields to lat and lon in telemetry screen

local function init()
  -- Called once when the script is loaded
	mlat =  getFieldInfo("lat").id    -- here we get the numerical id of the filed "lat"
	mlon =  getFieldInfo("lon").id    
	mGSpd =  getFieldInfo("GSpd").id    
	mHdg =  getFieldInfo("Hdg").id 
	mGAlt =  getFieldInfo("GAlt").id 	
end

local function run()
  -- Called periodically while the Special Function switch is on
--{"OTA":0, "lat":435128245, "lon":14970095, "GSpd":0, "Hdg":193.33999633789, "GAlt":183.60000610352}
	serialSetPower(1, 1) -- setPower port AUX2 on --> serialSetPower(port_nr, value)
	serialWrite('{"OTA": 0, "lat":'..getValue(mlat)..', "lon":'..getValue(mlon)..', "GSpd":'..getValue(mGSpd)..', "Hdg":'..getValue(mHdg)..', "GAlt":'..getValue(mGAlt)..'}\n')
end

local function background()
  -- Called periodically while the Special Function switch is off
end

return { run=run, background=background, init=init }

frankiearzu, 2023-09-21:
Just got a clarification from Msterns... 1E7 means 1,000,000...   my confusion! when i see a number with A..E automatically my mind thing of Hex!  
Check the new firmware.  I think we will be close..  if the decimal "," is not in the right place, we just need to adapt to do a different division.

f2knpw, 2023-09-21:
> Just got a clarification from Msterns... 1E7 means 1,000,000... my confusion! when i see a number with A..E automatically my mind thing of Hex! Check the new firmware. I think we will be close.. if the decimal "," is not in the right place, we just need to adapt to do a different division.

I have just uploaded the firmware and here is what I get 

[image]
 Still 2 GPS fields one with lon the other with lat
But not yet perfect.

f2knpw, 2023-09-21:
> dude, getValue(mGPS.id).lat

[image]

mGPS should the "id" ???

	mGPS =  getFieldInfo("GPS").id    -- here we get the numerical id of the field "GPS"  

and then 

.getValue(mGPS).lat..',

frankiearzu, 2023-09-21:
can you change the GPS format to NMEA, and can you check if the decimal is close to where it should be.  With NMEA will be close to the decimal value...  I know that you don't want to disclose your location.   I will check about the 2 sensors.. should be only one... but at least progress in getting the right format.

mha1, 2023-09-21:
read the documentation. sensor GPS returns a table. You need to fetch the table and then use the table to get to table entries latitude and longitude.

value = getValue(getFieldInfo("GPS").id) -- fetch the data table for sensor GPS. value now holds the table(!)
lat = value.lat -- fetch the table entry for latitude
lon = value.lon -- fetch the table entry for longitude

f2knpw, 2023-09-21:
> can you change the GPS format to NMEA (Systm Settings->GPS), and can you check if the decimal is close to where it should be. With NMEA will be close to the decimal value... I know that you don't want to disclose your location. I will check about the 2 sensors.. should be only one... but at least progress in getting the right format. Just double checking, you are using ibus, right?

I have switched to NMEA. 

But I get exactly the same values as when it was in DMS.

SO I get DMS values into two new fields.

I did cleared the telemetry and discovered again --> same thing.

frankiearzu, 2023-09-21:
Just did a clean build... should be only 1 sensor, since i deleted/commented the sencond from the table. 
[image]

New firmware( different name... #2 at the end) 
https://github.com/frankiearzu/DSMTools/blob/main/EdgeTx_Firmware/tx16s-edgetx-v2.9.0-ibus-gps2.bin

f2knpw, 2023-09-21:
> read the documentation. sensor GPS returns a table. You need to fetch the table and then use the table to get to table entries latitude and longitude.
>
> value = getValue(getFieldInfo("GPS").id) -- [...]

Ok thanks again !
I will try.

As I said lua language is still a mystery for me ... I don't even know what is a table !

f2knpw, 2023-09-21:
> Just did a clean build... should be only 1 sensor, since i deleted/commented the sencond from the table. [image]
> 
> New firmware( different name... #2 at the end) https://github.com/frankiearzu/DSMTools/blob/main/EdgeTx_Firmware/tx16s-edgetx-v2.9.0-ibus-gps2.bin

I downloaded this GPS2 firmware and tested.
Same thing : nothing has changed

I get the right values but expressed in Degree Minute.decimal minute
Still two fields as it was on my previous picture

mha1, 2023-09-21:
https://www.lua.org/pil/2.5.html

have you considered reading a book or studying the language online? I think you should. I also believe think you should transfer questions on how to implement your project to e.g. https://discord.com/channels/839849772864503828/842693696629899274 or https://discord.com/channels/839849772864503828/1051165136473313290

We are spamming a bug tracker ...

f2knpw, 2023-09-21:
So with NMEA it gives DD MM.MMMM
with DMS it gives DD MM SS.SSSS

frankiearzu, 2023-09-21:
Ok, did a change... the sensor is named "GPS3" instead of "GPS".
https://github.com/frankiearzu/DSMTools/blob/main/EdgeTx_Firmware/tx16s-edgetx-v2.9.0-ibus-gps3.bin

Just to double check. to update your firmware, you copy the file into the /FIRMWARE directory, turn off the radio, start the radio in bootloader mode (T4,T1 inwards while powering on), in bootloader you choose the right firmware file to update your radio.

f2knpw, 2023-09-21:
> We are spamming a bug tracker ...

Sorry for that !

f2knpw, 2023-09-21:
> UPDATED: Just saw you image...is using differen instance# V4 overrides the instance to 0. https://github.com/frankiearzu/DSMTools/blob/main/EdgeTx_Firmware/tx16s-edgetx-v2.9.0-ibus-gps4.bin
> 
> usually instance is when you have multiple sensors of the same type.. unlikely that you will have 2 GPS sensorts, and that is to differenciate. I think that could be a bug in oXs... the instance ID in the package is the device.. if you only have one GPS sensor, all the InstanceIDs should be the same.. as you can see in the image, the instanceID is different per each sensor.

Not totally understood ! (was easier to read how to flash the firmware !)

But now the screen display looks good 👍 
[image]

when in NMEA I get lat lon in DD MM.mmmm

So still have to convert it to get decimal degrees. (DD.dddddddd)

frankiearzu, 2023-09-21:
Thats progress!! Thats the 2 formats  supported for display. DMS and NMEA (DD MM.mmmm)
Internally is the same as the raw data transmitted by the RX.  To be honest, not sure what is the internal format, i think is decimal degress but without the decimal point (sotoed as integer).

**Just one last question:**
If you copy/paste that GPS location, does it matches your location?
If so, i think we are good on the EdgeTX side, and  will create the pull request to check-in the code into the Main EdgeTX branch.

Asked MSterns about the instanceID.. i think all the sensors comming from the same device (One GPS device in this case) should have the same instanceID, but if you see in your screenshot.. they are all different.

f2knpw, 2023-09-21:
Yes it's the right location !

I have now to figure out how to read this table with lua script. Tomorrow will be another day !
Thanks for your help

frankiearzu, 2023-09-21:
My sugestion for LUA will be to make a simple tool script who displays the Lat/Log on the screen.  You can start with Companion so that you don't need to debug on your radio.  Companion has a Sensor simulation, and i think GPS is one that it can simulates.
Once it works in Companion, you can try in your radio.   
Check this sample:
https://luadoc.edgetx.org/part_i_-_script_type_overview/one-time_scripts   Inside the "run()" function, use lcd.clear() and lcd.writeText(x,y,text,0).  Start by displaying "Hello World".. after that, get the sensor values and display the data you need

f2knpw, 2023-09-22:
Hi again,
 I believe that there is a side effect with the "v4" firmware.

In the system/hardware menu when scrolling to the AUX1 and AUX2 serial ports, the AUX2 has disapeared...

V2.9 screen
[image]

and now with the 2.9 #4

[image]

f2knpw, 2023-09-22:
> well, look at my code, I'll trust you figure it out. If not try this, guaranteed to be not tested:
> 
> ```
> local mLat = ""
> local mLon = ""
> 
> local fi = getFieldInfo("GPS")
> 
> if (fi ~= nil) then
>   mLat = getValue(fi.id).lat
>   mLon = getValue(fi.id).lon
> end
> ```

Hi,

I have tested also this code into V2.9 of the firmware.

It doesn't work probably because .lat and .lon does not exist for this "GPS" sensor
[image]

The code almost work like this (accessing directly "GPS")
[image]

But I only get the longitude and can't get the latitude even when trying to fetch the field.id + 1
(as latitude is the next field after longitude)

(BTW : I go on writing into this thread as it is IMHO related to the initial bug.)

f2knpw, 2023-09-22:
so to summarize a little

I suspect that the 2 "GPS" fields are totally distinct and not a "bundle".
But as they do have the same "GPS" name I can only access one of the them (first one it seems)

If I rename them with 2 diffferent names (lat and lon) then I can access both.

[image]

mha1, 2023-09-22:
well, this is because the current implementation of your selected telemetry protocol doesn't deliver the proper "bundled" GPS position. It works for other protocols. 

[image]

f2knpw, 2023-09-22:
I do have a TX16s

I have tried to find the "id" of the two GPS fields
lon is 264
lat is 267

forcing the second id gives good results...But not very handy!

[image]

pfeerick, 2023-09-22:
In this instance, reading the table will only work with what you refer to as the v4 firmware (where the two values are merged into the one sensor line) ... otherwise, you need to use getValue with the two different fields like you show above.

f2knpw, 2023-09-22:
I have this code working but I don't know why id of the "next" sensor is id +3

[image]

I have tried to delete the telemetry and rediscover ... seems to work all the time !

BTW : currently I can't use the V4 as it has removed the AUX2 port on my radio !

f2knpw, 2023-09-22:
> In this instance, reading the table will only work with what you refer to as the v4 firmware (where the two values are merged into the one sensor line) ... otherwise, you need to use getValue with the two different fields like you show above. Assuming no issues, #4080 will get merged, and it may get included to 2.9.1

But there is still an issue --> AUX2 disappears from my screen. (see above)

pfeerick, 2023-09-22:
That is more likely related to something else in Frankie's build environment... the changes in #4080 should not be causing that. If you can check it, since you have a test setup already... back up your radio SD card MODELS and RADIO folders first, and try the firmware from that PR out via the Check tab of that PR (here's the direct link to the zip file with the firmwares: https://github.com/EdgeTX/edgetx/suites/16432097766/artifacts/939003405).

mha1, 2023-09-22:
Please try the attached firmware on your tx16s, delete all sensors, rediscover and change your LUA to use

```
local mLat = ""
local mLon = ""

local fi = getFieldInfo("GPS")

if (fi ~= nil) then
  mLat = getValue(fi.id).lat
  mLon = getValue(fi.id).lon
end 
```

[firmware.zip](https://github.com/EdgeTX/edgetx/files/12699966/firmware.zip)

mha1, 2023-09-22:
sorry, my mistake, sent wrong tx16s firmware. Flash this one: [firmware.zip](https://github.com/EdgeTX/edgetx/files/12700099/firmware.zip)

f2knpw, 2023-09-22:
> Please try the attached firmware on your tx16s, delete all sensors, rediscover and change your LUA to use
>
> ```
> local mLat = ""
> local mLon = ""
>
> local fi = getFieldInfo("GPS")
>
> if (fi ~= nil) then [...]

I have tried. Doesn't work...
Still two fields named "GPS". And lua script fails to execute when .lat .lon

[image]

f2knpw, 2023-09-22:
Argh... didn't se your last message ! I try again

mha1, 2023-09-22:
sorry

mha1, 2023-09-22:
you have AFHDS2A as protocol, correct?

f2knpw, 2023-09-22:
Yep AFHDS-2A with FSIA6B receiver.

Tested your last firmware. Kinda works ...
[image]

I do get a new "bundled" GPS but with values at 0
And still latitude into another field named 80

mha1, 2023-09-22:
Probably my bad: [firmware2.zip](https://github.com/EdgeTX/edgetx/files/12700284/firmware2.zip)

f2knpw, 2023-09-22:
> That is more likely related to something else in Frankie's build environment... the changes in #4080 should not be causing that. If you can check it, since you have a test setup already... back up [...]

Ok just tested your firmware.
It seems to work fine
I have again the AUX2 into the hardware
And now a nice GPS bundle

[image]

Still have to check if I can read this "bundle" !

mha1, 2023-09-22:
no more sensor 80 too, correct?

f2knpw, 2023-09-22:
That's absolutly fantastic !

I just had to remove the division by 10000000 into my code and it worked !

[image]

So @pfeerick's latest firmware. With @mha1 lua's code

[image]

And my little remoteID beacon is working again !
[image]

Interestingly there is no conversion now to apply. And lat lon values are in decimal degrees (which is cool)
[image]

So I will keep this firmware waiting for the 2.9.1 official release.

**MANY** thanks for all the efforts made for this little bug !

mha1, 2023-09-22:
The number of satellites seems odd. 112?

f2knpw, 2023-09-22:
Yes you are right. They increase from 0 to 4 then go to 105 then 112.

I assume that the first digit is the fix (0 no fix, >100 fix ) !

[11 more replies]

PR #4080 (2023-09-26): fix(ibus): Separate lat long sensors merged to single GPS sensor
Update code to populate single sensor GPS with Lat/Long together

Fixes #4062
Code is safe to be released with 2.9.1

Summary of changes:
This is for FlySky_IBUS protocol.
The original code was poulating GPS latitude and longitude in two instances of the "GPS" sensor. The bug fix asked to give them a different name (GLat.GLon), but the standard in many other protocols in EdgeTX is to have both in a single sensor.
It is displayed correctly in the UI and Widgets. Also the logs will inteact properly with goggle maps.

**Testing:**
S [... cut]
closed 2023-09-26 by pfeerick