EdgeTX/edgetx #4069: MPM issue with Horus X10 S Express and Edgetx 2.9

Barstock, 2023-09-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After upgrading from Edgetx 2.8.5 to 2.9 I experienced the following:
1 Unable to maintain a link with Flysky FS-IA10B receiver. The connection last approximately 3 - 5 seconds  then the link terminates and the External Module is not detected by the transmitter.
2 Unable to complete a new bind to the Flysky receiver or a Hitec Optima.
3 Am able to connect to and operate previously bound Hitec Optima and Graupner receivers.
Nb. The internal module was off during these operations.

### Expected Behavior

Prior to the upgrade I’ve experienced no issues with the MPM module in this transmitter through Opentx and Edgetx earlier versions.
I’ve rolled back to Edgetx 2.8.5 and the system operates normally

### Steps To Reproduce

This would entail re flashing to 2.9

### Version

2.9.0

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Windows

### Anything else?

Transmitter Module ISRM S X10E
Status 1.1.0/2.1.9 FCC
MPM IrangeX 4 in 1 V1.3.3.20
 I have two of these modules and both have operated without flaw prior to the upgrade.

raphaelcoeffic, 2023-09-20:
I have only the 8 channel receiver. Is that specific to that 10 channel receiver?

raphaelcoeffic, 2023-09-20:
Ok, I tried all sorts of combinations, including updating the internal module to FW v2.1.9, I cannot seem to be able to reproduce anything with the FS-iA6B receiver.

> The connection last approximately 3 - 5 seconds then the link terminates and the External Module is not detected by the transmitter

What do you mean exactly with "external module is not detected"? Is that the module status line telling you it has no telemetry?

Barstock, 2023-09-20:
Good morning,

Seems like I had a gremlin in the system.

I've just re flashed the transmitter to Edgetx 2.9 and the system is
functioning fine.

Unfortunately I didn't take a photo of the Module Status page when it
malfunctioned, the version wasn't shown but words to the effect "no serial
..... detected".

Thanks for your prompt response and I'll keep you posted,

I'm satisfied to consider the issue closed

Regards

closed 2023-09-20 by raphaelcoeffic