EdgeTX/edgetx #4073: QX7 restarts or glitching on 2.9.0 , no problem on 2.8.5

zigg1312, 2023-09-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Occasional restarts even when idle. Not full reboot.  
background.  
Upgrade to 2.9.0 and system rebooting fully occasionally. 
Downgrade to 2.8.5 and was able to use without incident for multiple hours in liftoff using internal xjt and frsky rx (XSM-SIM) 
Facebook group suggested clearing sdcard which I replaced and dl 2.9 sdcard contents and reloading models and settings from OPENXT backup 
reflashed to 2.9.0 and will glitch - screen blanking and speaker click and loss of connection but not full reboot like previously 

### Expected Behavior

No - expect it to work as well as 2.8.5

### Steps To Reproduce

Just sitting idle it restarts about every minute

### Version

2.9.0

### Transmitter

FrSky Q X7 / Q X7S (ACCST)

### Anything else?

Willing to do anything to debug - just need instructions. However will be reflashing to 2.8.5 to escape this problem, but can reflash if requested. 
Have done the crossfire mod to transmitter , but everything in 2.8.5 works as expected for multiple hours , just 2.9.0 is problematic.

Many thanks and great work !

zigg1312, 2023-09-20:
Just noticed flashing exclamation mark in top left corner (!) - cannot find explanation of what this is signalling

philmoz, 2023-09-21:
Can you zip up you MODELS and RADIO folders plus any Lua scripts being used and post them here please.

philmoz, 2023-09-21:
> Just noticed flashing exclamation mark in top left corner (!) - cannot find explanation of what this is signalling

This means the radio crashed and rebooted into emergency mode.

zigg1312, 2023-09-21:
[modelsradioscripts.zip](https://github.com/EdgeTX/edgetx/files/12682021/modelsradioscripts.zip)

Thanks Philmoz!

zigg1312, 2023-09-21:
model i was using is called sim uses the xjt internal d16 with telemetry but not discovered or use sensors, but all models probably all use xjt except for one with r9

philmoz, 2023-09-22:
Can you please try the attached model file on your radio.

Rename the file to model09.yml (from model09.txt) and replace the model file on your SD card.

[model09.txt](https://github.com/EdgeTX/edgetx/files/12696408/model09.txt)

zigg1312, 2023-09-22:
Tried that model, no change. Interesting development - I bound using irange 4in1 instead of XJT and internal module off and everything fine, no crash or emergency mode. which suggests differences in radio communication between versions

philmoz, 2023-09-22:
> Tried that model, no change. Interesting development - I bound using irange 4in1 instead of XJT and internal module off and everything fine, no crash or emergency mode. which suggests differences in radio communication between versions

Very strange. Your model09.yml file crashes my X7S consistently; but it is ok with the changes I made.

zigg1312, 2023-09-22:
maybe flash to 2.8.5 to see if it crashes you on that version?  Anyway will create a new model in 2.9 and see where I get. with that - I dont mind if I have to recreate all models as I am just getting back into it after giving up a few years ago when opentx was at 2.2 something

philmoz, 2023-09-22:
There's definitely something crashing in the code; but at least now we can reproduce it (at least I can).
That's a big step towards finding and fixing the cause.

raphaelcoeffic, 2023-09-22:
It seems the actual issue is the SBUS trainer which setup in this model (see #4086)

raphaelcoeffic, 2023-09-22:
@zigg1312 @philmoz would you mind trying this firmware?

[firmware.zip](https://github.com/EdgeTX/edgetx/files/12698013/firmware.zip)

philmoz, 2023-09-22:
I tried a build of main (2.10) with your patch and no crashes so far (previously it would crash very quickly).

PR #4085 (2023-09-23): fix: prevent enabling USART IRQ if not supported
Fixes #4073
commit 96c773af (2023-09-23): fix: Prevent enabling USART IRQ if not supported (#4085)

Fixes #4073
PR #4086 (2023-09-23): fix: prevent enabling USART IRQ if not supported
Fixes #4073

In the issue at hand, the USART IRQ is enabled whereby the hardware definition does not provide a valid `IRQn` for this USART:
```c++
static const stm32_usart_t sbus_trainer_USART = {
  .USARTx = TRAINER_MODULE_SBUS_USART,
  .GPIOx = TRAINER_MODULE_SBUS_GPIO,
  .GPIO_Pin = TRAINER_MODULE_SBUS_GPIO_PIN,
  .IRQn = (IRQn_Type)-1,
  .IRQ_Prio = 0,
  .txDMA = nullptr,
  .txDMA_Stream = 0,
  .txDMA_Channel = 0,
  .rxDMA = TRAINER_MODULE_SBUS_DMA,
  .rxDMA_Stream = TRAINER_MODULE_SBUS_DMA_STREAM_LL,
  .rxDMA_Channel = TRAINER_MODULE_SBUS_D [... cut]
commit 60ccd852 (2023-09-23): fix: Prevent enabling USART IRQ if not supported (#4086)

Fixes #4073
closed 2023-09-23 by pfeerick