EdgeTX/edgetx #4099: Companion - Calibration is not read correctly

TheIsotopes, 2023-09-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

with latest Companion nightly `0aff002c` the calibration is not read correctly

see the Rudder on screenshot - S1 and S2 are ok, these have been deactivated

[image]

### Expected Behavior

Correct calibration values when reading out the transmitter

### Steps To Reproduce

...

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster Zorro

### Operating System (OS)

Windows

### OS Version

11

pfeerick, 2023-09-24:
@TheIsotopes Can you double-check something for me on another issue... does the edit button on the toolbar (and/or the context menu) work for you? If not, I need to open an issue for that, as can only edit by double-clicking on the model atm. 🤪 

[image]

TheIsotopes, 2023-09-24:
@pfeerick You are right, the button has no function.

elecpower, 2023-10-16:
Companion Nightlies are broken again with the latest adc refactor on the radio side.

pfeerick, 2023-10-16:
Now that you're back, https://github.com/EdgeTX/edgetx/pull/3870 is going to be merged in the next couple of days, so expect more stuff to break 😆 This is the last expect breakage for this cycle.

PR #4406 (2024-02-04): refactor(cpn): Support JSON radio hardware definitions
Summary of changes:
- remove most of the 1st attempt
- move from static board configurations to use radio definitions json files. This also has the advantage of keeping radio sims, firmware and Companion in sync
- group separate structs for same item eg pots into single structs
- maintain support for reading legacy radio settings configuration formats eg bin, otx, etx and convert all formats inc yml as necessary
- yaml encode and decode
- IMPORTANT: 'cmake --build . --target native' and 'cmake --build native --target hardware_defs' MUST be [... cut]
closed 2024-02-04 by pfeerick