EdgeTX/edgetx #4106: No Fail-Safe Setting in Companion

Hartinger, 2023-09-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

With Companion 2.9.0 the settings for the internal RF module do no longer contain any fail-safe settings.

[image]

### Expected Behavior

Fail-safe can be configured as part of the settings for the internal module, as per Companion 2.8.5.

[image]

### Steps To Reproduce

1. Start Companion
2. Load an etx file

### Version

2.9.0

### Transmitter

FrSky X9D+2019, FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Windows

### OS Version

Windows 11

msun12000, 2023-09-29:
I can confirm exactly the same issue with my Transmitter X9 Lite S. The failsafe settings are accessible on the TX and the settings are preserved when writing to and then reading from Companion 2.9.0

pfeerick, 2023-10-18:
The missing USB joystick config was a todo from #2277 that I missed when it was taken over into #3111

PR #4220 (2023-10-19): fix(cpn): Add back failsafe settings for protocols that support
Fixes #4106
closed 2023-10-19 by pfeerick