EdgeTX/edgetx #4107: Timer run

franz4949, 2023-09-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

In setup of Timer if selected as source GAS or GAS% it starts/runs on basis of GAS MIXER output. As a result of this the timer will even run when gas stick is moved and motor protection switch is still active (overwrite gas channel with -100%). Timer should however not run as long as gas channel output ist still -100%.

### Describe the solution you'd like

Please modify basis of timer and take GAS CHANNEL output.

pfeerick, 2023-09-26:
Throttle, Throttle % and Throttle Start (or GAS in the German builds) is tied to the Throttle Stick directly, not the mixer output, or even outputs. You can see this quite clearly in action if you use the more preferred throttle cut of a replace mixer line (which uses an offset to push the max value to min)

[image]

[image]

... the timer behaviour will still respond to the throttle stick position contrary to the mixer. 

Perhaps use the timers "Switch" option to only enable the timer when you don't have the throttle cut enabled?

Alternately, perhaps we need to add a Throttle Cut switch option to the Model Setup -> Throttle page, that allows a throttle cut switch to be setup up properly, and thus the rest of the firmware can work with it properly.

franz4949, 2023-09-26:
I found the way to do it analogue to OTx there's another option for Throttle Control, there I can select the source for throttle timer where I can select e.g. also a channel output as source.

So I close this issue, thanks for your kind reply.

closed 2023-09-26 by franz4949