EdgeTX/edgetx #4114: Touch Screen Mod for the Horus X10 / X10S

MRC3742, 2023-09-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No

### Describe the solution you'd like

With the availability to mod the other color LCD radios and add the TX16S touch panel (Horus X12, Jumper T16, and T18)
I'm looking to add the touch panel to the X10 series of transmitters.

It appears that the same mod as done for the Jumper T16 should work on the X10 / X10S.
The firmware would need to be modified to allow these radios the ability for an added touch panel.

I have already attempted this firmware and hardware mod on an X10 (non Express) and have made the changes as was done to add touch to the Jumper T16 on PR #2060 
The radio will start with just the firmware changes and it also shows (although not working) Touch GT911 on the Hardware Analogs screen.
After the hardware mod is done, as on the T16, the radio starts with only "Wrong PCB Detected" on screen and will shut off normally with power button.
Changing back the SCL, SDA lines modded from pull up (3.3V) back to pull down (Ground) the TX will now start normal.
I'm missing something that is causing Pin 97 (PH.07) and pin 98 (PH.08) either floating or pulled up to cause this error.

Any advice?
Thanks - Rich

pfeerick, 2023-09-27:
This sounds like you're tripping the hardware detection that would normally be used to detect PCB revisions. i.e. when a manufacturer makes a revised version of a board that has changes the software needs to know about, they usually set an otherwise unused MCU pin to a known state which the firmware can detect the state of, so it can adapt its behaviour to suit the board. 

For the X10, it looks like H7/H8 are those indicator pins! :-O
https://github.com/EdgeTX/edgetx/blob/f7b676fcad8056f1df2e9d70359cec10618a60c4/radio/src/targets/horus/hal.h#L397-L402

I don't immediately see any other reference to H.07/H.08 being in use for that board, so you could possibly try bypassing the check here by setting it to true for your target?
https://github.com/EdgeTX/edgetx/blob/f7b676fcad8056f1df2e9d70359cec10618a60c4/radio/src/targets/horus/board.h#L87-L94

MRC3742, 2023-09-27:
Success!  I now have the touch screen working on my X10

Thank you for that information.
I knew it was some sort of board check but was not able to find it in the code.
The bypass on line 93 set to "true" works but is probably not the proper way to add this mod to the codebase?

Thank You Again - Rich

[image]
[image]

pfeerick, 2023-09-27:
Not as a "proper" fix I don't think. 

If you get a chance, can you see if swapping `PCBREV_X10_STD` for `PCBREV_X10_EXPRESS` on ln93 allows it to boot without issue...  (meaning the code simply can no longer differentiate between the X10 and X10 Express). If so, may have to consider a flag that disables the check in order to support moding the X10 (and probably X10 Express, I guess).

MRC3742, 2023-09-27:
@pfeerick 
Building using `PCBREV_X10_EXPRESS` on ln93 also allows the radio to start and work with touch control.
Could a check be added after the current one to see if those pins are pulled up? then touch support could be allowed?

I also have a Horus X10S-EXPRESS that I can use to check the MCU pin connections.
The resistor "R1" used to pull down pin 98 (PH.08) is not populated on my X10S-EXPRESS board.
Would this be used for the check to distinguish between the Standard or EXPRESS ?
[image]
X10S/EXPRESS Mainboard MCU with R1 not populated.

Thanks - Rich

MRC3742, 2023-09-27:
@raphaelcoeffic 
Although AUX1 would be easier to hook up to originally, I was hoping to leave this available to use for other options.
Such as the External Access Mod connection point on the X10 Standard Tx, which is already available via AUX1 choices.

Thanks - Rich

raphaelcoeffic, 2023-09-28:
> Such as the External Access Mod connection point on the X10 Standard Tx, which is already available via AUX1 choices.

Right, makes great sense!

MRC3742, 2023-10-02:
I think I've managed to allow the PCBREV check on the X10 Standard (NON EXPRESS) mainboard to succeed as intended.
In addition to the Touch Screen Mod you would need to add a 1K pulldown resistor to ground on unused MCU pin 55 (PA6)
The X10 / X10S EXPRESS radios do not need this additional pulldown as they will already pass the check after installing the Mod.

I've added PR #4151 for your review and suggestions.
Thanks - Rich

MRC3742, 2023-12-22:
Information for Touch Panel update set (uses your current LCD screen) or complete new IPS LCD with Touch Panel assembly.

This Mod adds touch screen capability to the Horus X10 series radios with just a "TX16S Touch Panel update set".
Approximately $10.00 USD plus S&H
Currently available in US here [https://graysonhobby.com/radiomaster-tx16s-trouch-panel-update-set.html](url)

A better option is to use a TX16S "Replacement **IPS** LCD Screen and Touch Panel assembly".
This IPS screen is clearer and has a much better viewing angle than the original X10 LCD screen. (Very noticeable improvement)
Approximately $30.00 USD plus S&H
Currently available in US here [https://www.buddyrc.com/products/radiomaster-tx16s-lcd-touch-panel](url)

Rich

MRC3742, 2023-12-23:
Code for addition to add Touch Screen Mod for the Horus X10 Series has been merged into 2.9 and main branches.

Issue Closed

closed 2023-12-23 by MRC3742

NiklasVoigt, 2024-02-01:
@MRC3742 can you tell us which pins you used on the main processor? maybe write a little guide?

MRC3742, 2024-02-02:
As referenced in the first post of this issue, I just followed a guide for installing touch on the Jumper T16 that was made by Stinky80. [how_to_add_toch_in_T16.pdf](https://github.com/EdgeTX/edgetx/files/8928271/how_to_add_toch_in_T16.pdf)

If modding the X10 NON EXPRESS, you also need the 1K resistor mod listed above.
[Resistor mod for NON EXPRESS models post](https://github.com/EdgeTX/edgetx/issues/4114#issuecomment-1742439635)

The only other problem is the touch panel adds approx. 1mm thickness and you need to remove that much material from the ribs on the front of plastic piece that holds the screen in place. I used a dremel and a sanding wheel to remove the same amount as added by the new screen thickness.

This picture posted by Stinky80 can be used as a reference and shows pins needed.
The Jumper has the same basic layout as the X10 and X10EXPRESS.

[image]

nenno123, 2024-11-10:
hi... I want to make this Mod... I have a X10S Express. Can someone please check this photo.

[image]

Is this correct?
I have to remove R2 (R1 isn't populated). The rest should be clear... 

Regards

Mike-HRC, 2025-01-12:
So if I add this mod, it will now work with new releases of EdgeTx with the generic new releases of ETX?

nenno123, 2025-08-08:
I can confirm that the TouchMod is still working perfectly with the newest EdgeTX Version

Giorg10, 2025-08-08:
thank you nenno. Re-inspected the hw and it turned out I was using my fpc connector on the wrong side. Flipped the cable, inverted all the connections and everything is now working!

PR #4151 (2023-12-23): Allow Touch Screen Mod on Horus X10 Transmitters

Fixes #4114 

Changes to allow the Touch Screen Mod for the Horus X10 Series of Transmitters.
This Mod requires some soldering directly to the MCU processor pins as no other connection points are available! 

This is very similar to the already documented Jumper T16 Mod (PR #2060) that has been successfully added to EdgeTX.
The X10 / X10S Standard (NOT EXPRESS) radios need an additional pull down 1K resistor attached to MCU pin 55 (PA6)
This will allow the firmware to pass the PCBREV check as intended to distinguish between Standard or EXPRESS  [... cut]