EdgeTX/edgetx #4118: Hash Marks for Configuration Sliders

jmxp69, 2023-09-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

In radio settings for sound and haptic there are a number of sliders. The volume slider looks almost perfectly analog with very granular control. The beep volume slider only has four steps. Beep length four steps, Wav volume four steps. Background volume four steps. Haptic length and strength are four steps. 

The differences in these steps indicate a difference in the touch motion required to advance the slider that is not necessarily intuitive.

### Describe the solution you'd like

Ideally, could all of the sliders be made more analog like the volume slider?
If that is not possible, could hash marks/tick marks be added underneath the sliders indicating how many steps are available per slider so the user knows about how far touch needs to go to activate a desired step?

### Describe alternatives you've considered

Mixing whiskey and prescription drugs--but my Dr. advises against it.

### Additional context

This is the best opensource crew on Github!

philmoz, 2023-09-27:
Tick marks are possible:
[image]

@JimB40 - Does this look ok?

jmxp69, 2023-09-27:
That's definitely the idea. Maybe dots would look better/less busy?? You're the graphics guy...but yes, that's the idea. Gives a visual indication of slider stops.

philmoz, 2023-09-28:
I tried a dotted line on my TX16S; but it was very hard to see.
A dashed line was not too bad:
[image]

Removed the tick on the volume slider here as well. Personally I prefer having them; but I will defer to @JimB40 on which is better from a UX perspective.

JimB40, 2023-09-28:
@philmoz I will write further comments in #4119, considering what is mentioned here by @jmxp69

PR #4125 (2023-10-16): feat(color): Slider styling and tick marks.
Fixes #4118 and #4119
closed 2023-10-16 by pfeerick