EdgeTX/edgetx #4119: Fix UI Slider Styling

JimB40, 2023-09-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Now UI slider is bit overwhelming in comparison to previous styling
[image]

### Describe the solution you'd like

[image]

Normal State

1. Change TRACK height back to 8px
2. Increase KNOB border to 2px as other elements

Focused State
TRACK color and KNOB border color changes to FOCUS color

Edited state
TRACK color changes to EDIT color while KNOB border color stays in FOCUS color to give separation

JimB40, 2023-09-28:
[image]

Alternative version of EDIT state.
IMHO changing track color is better. What you think @philmoz?

philmoz, 2023-09-28:
Probably worth combining this with #4118 - can you please provide feedback on the tick mark concept.

philmoz, 2023-09-28:
Something else to consider - the slider knob does not have to be round.

[image]

JimB40, 2023-09-28:
Indeed in fact in initial design #3258 I've made slider knob rectagular with round edges, leaving circle for Switch only.
[image]
Should be slightly wider.

Are those tick marks LVGL standard feature?

JimB40, 2023-09-28:
3 different styles as propsal
[image]

philmoz, 2023-09-28:
The tick marks are being drawn manually - not an LVGL feature.

The knob can be either a rectangle with round corners or a circle - so the 'Beep length' option in the image above is only one that could be done.

JimB40, 2023-09-28:
I think dots as John mentioned look more clean
[image]

and there is certain level of granulatity at which dots shoul not be displayed.
To my eye more than 15-16 levels.. but this depends on track size.

philmoz, 2023-09-28:
I can't draw between the slider and knob - I can only draw custom lines underneath the whole slider object.

JimB40, 2023-09-28:
[image]
From what I can test, hotspot is a track not knob.
But hotspots are strangely defined.
It looks they follow Beep volume pattern rather than Beep lenght (which is proper UI method)

Can you confirm?
Have you got any control over hotspots?

philmoz, 2023-09-28:
> From what I can test, hotspot is a track not knob. But hotspots are strangely defined. It looks they follow Beep volume pattern rather than Beep lenght (which is proper UI method)
> 
> Can you confirm? Have you got any control over hotspots?

Sorry, I don't understand the question.

JimB40, 2023-09-28:
> > From what I can test, hotspot is a track not knob. But hotspots are strangely defined. It looks they follow Beep volume pattern rather than Beep lenght (which is proper UI method)
> > Can you confirm? Have you got any control over hotspots?
> 
> Sorry, I don't understand the question.

Try to slide you finger on the track and observe when knob changes position. It follows first hotspot layout (marked orange) rather then desired second layout.

JimB40, 2023-09-28:
[image]
Conept with 2px dots over track

philmoz, 2023-09-28:
> Try to slide your finger on the track and observe when knob changes position. It follows first hotspot layout (marked orange) rather then desired second layout.

That's all controlled by LVGL.

philmoz, 2023-09-28:
Aligning the dots to the center of the knob looks like this. I think it looks better without the end point dots.

[image]

JimB40, 2023-09-28:
Yes, end-points are obvious

JimB40, 2023-09-28:
> > Try to slide your finger on the track and observe when knob changes position. It follows first hotspot layout (marked orange) rather then desired second layout.
> 
> That's all controlled by LVGL.

Then we have to live with what LVGL guys coded.

closed 2023-10-16 by pfeerick