EdgeTX/edgetx #4120: Date resets when year reaches max

JimB40, 2023-09-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

https://youtu.be/nVDchdRzFIc

### Expected Behavior

FIx

### Steps To Reproduce

watch video

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

2.10 nightly

pfeerick, 2023-09-28:
"Fix" doesn't mean much... I presume you want the day and month to not be reset/change? This would probably need for the UI to be de-coupled from the hardware, as it is the RTC that you are setting live (as is the case for most fields that don't pop up) while editing the date that is saying the date is invalid, and resetting to 1-1-1970. This applies across the board - B&W and color.

philmoz, 2023-09-28:
Set an upper limit on the year value that does not overflow?

pfeerick, 2023-09-28:
Yeah, we might as well join the party... 2038 is a big year for everyone... 🤣 https://en.wikipedia.org/wiki/Year_2038_problem

PR #4124 (2023-10-03): fix(radio): Limit year range for RTC to prevent overflow
Fixes #4120
closed 2023-10-03 by pfeerick