EdgeTX/edgetx #4122: UI - MANAGE MODELS - rework

JimB40, 2023-09-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

[image]

### Describe the solution you'd like

1. Use standard LVGL button for sorting by name  
- button label  "A-Z" or "Z-A"

2. Button for sorting by date added
Frankly speaking I'd scratch it in favour of default 'Recent' group that will display 4-6 recently added  models

3. New button for changing selecting mode (Single label group / Multiply label groups 'AND')
This feature was requested by users.
We have bi-state button changing behaviour of label's group selection
a. Single - works like old groups, when you select label only models with this label are displayed
b. Multi - multiply labels can be selected - models that have all selected labels are displayed.   
- button labels: "Single", "Multi"

pfeerick, 2023-09-28:
Do you mean "opened" every time you said added? As for me, it would be a waste of screen space for a "recently added" label, I want to be able to access the models I most recently used ;) 

And I see you failed the "searched existing issues" test, and might want to look at / comment on rather than duplicating here:

1) https://github.com/EdgeTX/edgetx/issues/2802 
2) https://github.com/EdgeTX/edgetx/issues/3205

JimB40, 2023-09-28:
This is issue to summarize new features so thx for adding older ones.
I just repeated concept 'Recent' system label. :)
After reading it seems that logic behind this label should be "Recently used model"

ulfhedlund, 2023-12-29:
I'm not sure if this is the place to ask about a different alphabetical sort order. If not let me know and I can add an issue.

Currently (2.9.2) upper and lower case are sorted in two groups, i.e. first A-Z and then a-z, or reverse first z-a then Z-A. 

I think it would make more sense to sort alphabetically so that A-Z shows Aa, Bb ... Zz, and Z-A shows Zz, Yy ... Aa.

philmoz, 2023-12-30:
> I think it would make more sense to sort alphabetically so that A-Z shows Aa, Bb ... Zz, and Z-A shows Zz, Yy ... Aa.

I agree - the model sort should be case insensitive.

philmoz, 2024-09-10:
Is there anything here that is still an issue with the current manage models page, and settings?

JimB40, 2024-09-10:
dont think so. you changes & improvements went beyond orginal concept.

closed 2024-09-11 by pfeerick