EdgeTX/edgetx #4126: Vario volume sometimes louder for some beeps

mama99de, 2023-09-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

System Volume is set using global function and S1
[image]
Vario is activated using a Switch

Volume of the Vario is sometimes for some beeps (1-2 seconds) louder without changing system Volume.
Issue occours also using OpenTX 2.3.15 with my Xlite Pro, as well as with EdgeTX 2.9.0. It´s seems to be more frequent using EdgeTX. First i was thinking that the S1 Poti ist broken, but exact the same issue on my brand new RM Pocket, 

Sorry for bad quality of the video
Till 2 sec. it´s louder than normal, 2-10 sec. normal volume and at 11sec again lauder
https://www.youtube.com/watch?v=sY_aiqhhLhc

### Expected Behavior

Constant volume for vario

### Steps To Reproduce

Just activate vario and listen to Vario
Maybee also : 
System Volume is set using global function and S1
Vario is activated using a Switch

### Version

2.9.0

### Transmitter

FrSky X-Lite / S / Pro

### Operating System (OS)

Windows

3djc, 2023-10-04:
Act is ON only very shortly after stick, so your volume changes based (a bit randomly) on stick movement. Looks to me behaving as programmed

mama99de, 2023-10-04:
But why does the volume get louder and quieter? S1 Poti which is the input is not does not change.

3djc, 2023-10-04:
But your function gets some activated, sometimes not. Replace 'act' by 'ON'

mama99de, 2023-10-04:
While flying ACT is active all the time, there is no second i do not steer my planes. Will try it with ON, but looks still like an issue.
Volume must not change if Input does not change.

philmoz, 2023-10-04:
Go to the GF page in radio settings and watch GF1 (when it is set to Act).
It will be changing from white background (inactive) to yellow background (active) as you move or don't move the sticks.

When it goes inactive the volume will be controlled only by the 'Volume' slider in the Sound settings, not the S1 pot.

You need this GF to be active all the time which is why 3djc said to change it to 'ON'.

mama99de, 2023-10-04:
"When it goes inactive the volume will be controlled only by the 'Volume' slider in the Sound settings, not the S1 pot."
OK, got it, i thought only GF1 is not updated, when act is not active, didn´t know, that is will go back to System setting.

closed 2023-10-05 by 3djc