EdgeTX/edgetx #4127: UI - more compact MODEL/Timers

JimB40, 2023-09-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Curently Timers take 3 buttons in MODEL SETUP page

### Describe the solution you'd like

1. Make one button for timers
[image]

- if any of 3 timers is set "Timers" button get ACTIVE state
- 
2. New MODEL SETUP page
[image]

Navigation:
- Roller navigates through right-hand side UI elements (selected Timer option)
- PAGE> & PAGE< changes selected Timer (FOCUS frame) updating  right-hand side options

- if Timer is set ON corresponding button gets ACTIVE STATE

- this layout will allow to add more timers in future without expanding MODEL SETUP page.

pfeerick, 2024-09-10:
I like that alternative. If you can wrangle some way of showing which timers are not OFF in the tab bar... i.e. since color may not be a good option, perhaps some overlay icon? Unless it is perhaps done back on the button so you don't have to go into the next screen :thinking: Although that may limit things if more timers are added for some reason... :shrug:

philmoz, 2024-09-10:
Timer tab buttons can show in active color if active.
[image]

Timers button is highlighted if any timer is active.
[image]

We would lose visual indication on the setup page of which timers are active, only know that at least one is active.

JimB40, 2024-09-10:
@philmoz we should consider this as point in discusion about UI 3.0
Concept is about scratching top bar navigation in favour of centered and monochrome main menu.